Ultimaker/Cura #14656: Cannot create an installation file from source code

thecampbells384, 2023-02-24:
### Application Version

5.4.0

### Platform

Ubuntu 22.04

### Printer

any

### Reproduction steps

Following the instructions given in the previous issue: https://github.com/Ultimaker/Cura/issues/13568
```
pip install conan==1.56.0 --force
conan remove "*" -f
conan config install https://github.com/Ultimaker/conan-config.git
conan profile new default --detect --force

git clone git@github.com:Ultimaker/Cura.git
cd Cura
conan create . cura/5.4.0-alpha@me/testing --build=missing --update
cd ..
mkdir temp
cd temp
conan install cura/5.4.0-alpha@me/testing --build=missing --update -g VirtualPythonEnv -o cura:devtools=True -if cura_inst --json cura_inst/conan_install_info.json
source cura_inst/bin/activate
python cura_app.py
```

I have edited both Cura and CuraEngine to allow regional support extruder selection and ultimately want to create an install exe for this code, but for troubleshooting purposes a current clone of the Cura repo results in the same problem.

Additionally, in an attempt to package the code I subsequently ran:

```
pyinstaller cura_inst/UltiMaker-Cura.spec
python ./cura_inst/packaging/NSIS/create_windows_installer.py ./cura_inst/ ./dist/ cura.exe
```

If from the directory that Cura was cloned into I directly do the following then cura starts as expected
```
conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv
source cura_inst/bin/activate
python cura_inst/cura_app.py
```

### Actual results

When attempting to run cura_app.py I recieve an error that the ConsoleLogger plugin could not be found.

When attempting to create the windows installer I get an error that makensis can not be found

### Expected results

In the first instance I expected the cura app to start.

In the second instance I expected an install exe file to be produced

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

traceback from the failed running of cura_app.py
```
Traceback (most recent call last):
[... 212 lines cut ...]
  File "/home/fixed-command/.conan/data/cpython/3.10.4/_/_/package/41599e4c1f8b1b21e6c28e775ffeceed9a91f100/lib/python3.10/threading.py", line 966 in _bootstrap

Current thread 0x00007fabc0536740 (most recent call first):
  File "/home/fixed-command/cura/rapidiaCura5/temp/temp/cura_inst/lib/python3.10/site-packages/cura/CrashHandler.py", line 397 in _logInfoWidget
  File "/home/fixed-command/cura/rapidiaCura5/temp/temp/cura_inst/lib/python3.10/site-packages/cura/CrashHandler.py", line 182 in _createDialog
  File "/home/fixed-command/cura/rapidiaCura5/temp/temp/cura_inst/lib/python3.10/site-packages/cura/CrashHandler.py", line 86 in __init__
  File "/home/fixed-command/cura/rapidiaCura5/temp/temp/./cura_inst/cura_app.py", line 177 in exceptHook
```

output from attempted creation of the windows installer
```
Traceback (most recent call last):
  File "/home/fixed-command/cura/rapidiaCura5/temp/temp/./cura_inst/packaging/NSIS/create_windows_installer.py", line 82, in <module>
    build(args.dist_path)
  File "/home/fixed-command/cura/rapidiaCura5/temp/temp/./cura_inst/packaging/NSIS/create_windows_installer.py", line 72, in build
    subprocess.run(command)
  File "/home/fixed-command/.conan/data/cpython/3.10.4/_/_/package/41599e4c1f8b1b21e6c28e775ffeceed9a91f100/lib/python3.10/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/home/fixed-command/.conan/data/cpython/3.10.4/_/_/package/41599e4c1f8b1b21e6c28e775ffeceed9a91f100/lib/python3.10/subprocess.py", line 966, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/home/fixed-command/.conan/data/cpython/3.10.4/_/_/package/41599e4c1f8b1b21e6c28e775ffeceed9a91f100/lib/python3.10/subprocess.py", line 1842, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'makensis'

```

MariMakes, 2023-02-24:
Hey @thecampbells384,

😖 Sorry to hear you are stuck.

I'm only a fledgling member of the Build From Source Club 👑 
Before we get started, I have a few questions to narrow down the troubleshooting:

1. Can you make sure you have specifically Conan version 1.56.0?  
`pip install conan == 1.56.0 --force `
You can check the [Conan Documentation]( https://docs.conan.io/en/latest/installation.html ) for the latest version and installation instructions

2. Did you try clearing the Conan Cache? 
`conan remove "*" -f `

3. Can you make sure you have the latest config with this command?
`conan config install https://github.com/Ultimaker/conan-config.git`

4. Can you make sure you have a default profile?
`conan profile new default --detect --force`

5. With these steps you should have a clean slate and you can follow the steps on our wiki again. 
[You can find it here. ](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source) 

If you are stuck in any way, please reach out. We are happy to help out. 
The best way to do that is to provide us with your complete run-out log. 
Not just the part that fails, but all the lines generated after you've given your command. 
We need it to see your profiles and packages. 

Can you help me understand? What do you want to achieve by creating a Build?
Are you intending to share it with others or are you using it for research purposes?
I would love to know more about your project, maybe we can help.

thecampbells384, 2023-02-24:
Hi,
Thanks for the prompt reply. 

I did indeed already follow your advice in creating a clean slate, and following your wiki instructions to run the code from source works just fine. My problem is with packaging it into an install file. I tried following the instructions provided in this previous issue, but without success. https://github.com/Ultimaker/Cura/issues/13568

I have amended the cura code so that I can modify the support extruder regionally and we need to be able to distribute this code. If this was a feature that interests Ultimaker then I would be delighted to clean up our implementation a little and create a pull request, but in the mean time I (and it would seem others) would benifit from clearing up what the process looks like to package cura.

Complete outputs are as follows:

pip install conan==1.56.0 --force
```
Defaulting to user installation because normal site-packages is not writeable
Collecting conan==1.56.0
  Using cached conan-1.56.0-py3-none-any.whl
Collecting patch-ng<1.18,>=1.17.4
  Using cached patch_ng-1.17.4-py3-none-any.whl
Collecting bottle<0.13,>=0.12.8
  Using cached bottle-0.12.24-py3-none-any.whl (90 kB)
Collecting PyJWT<3.0.0,>=2.4.0
  Using cached PyJWT-2.6.0-py3-none-any.whl (20 kB)
Collecting colorama<0.5.0,>=0.3.3
  Using cached colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Collecting fasteners>=0.14.1
  Using cached fasteners-0.18-py3-none-any.whl (18 kB)
Collecting urllib3<1.27,>=1.26.6
  Using cached urllib3-1.26.14-py2.py3-none-any.whl (140 kB)
Collecting requests<3.0.0,>=2.25
  Using cached requests-2.28.2-py3-none-any.whl (62 kB)
Collecting pluginbase>=0.5
  Using cached pluginbase-1.0.1-py3-none-any.whl
Collecting six<=1.16.0,>=1.10.0
  Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting tqdm<5,>=4.28.1
  Using cached tqdm-4.64.1-py2.py3-none-any.whl (78 kB)
Collecting distro<=1.7.0,>=1.0.2
  Using cached distro-1.7.0-py3-none-any.whl (20 kB)
Collecting node-semver==0.6.1
  Using cached node_semver-0.6.1-py3-none-any.whl (10 kB)
Collecting python-dateutil<3,>=2.7.0
  Using cached python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
Collecting pygments<3.0,>=2.0
  Using cached Pygments-2.14.0-py3-none-any.whl (1.1 MB)
Collecting Jinja2<4.0.0,>=3.0
  Using cached Jinja2-3.1.2-py3-none-any.whl (133 kB)
Collecting PyYAML<=6.0,>=3.11
  Using cached PyYAML-6.0-cp310-cp310-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_12_x86_64.manylinux2010_x86_64.whl (682 kB)
Collecting MarkupSafe>=2.0
  Using cached MarkupSafe-2.1.2-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (25 kB)
Collecting idna<4,>=2.5
  Using cached idna-3.4-py3-none-any.whl (61 kB)
Collecting certifi>=2017.4.17
  Using cached certifi-2022.12.7-py3-none-any.whl (155 kB)
Collecting charset-normalizer<4,>=2
  Using cached charset_normalizer-3.0.1-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (198 kB)
Installing collected packages: pluginbase, patch-ng, node-semver, charset-normalizer, bottle, urllib3, tqdm, six, PyYAML, PyJWT, pygments, MarkupSafe, idna, fasteners, distro, colorama, certifi, requests, python-dateutil, Jinja2, conan
  Attempting uninstall: pluginbase
    Found existing installation: pluginbase 1.0.1
    Uninstalling pluginbase-1.0.1:
      Successfully uninstalled pluginbase-1.0.1
  Attempting uninstall: patch-ng
    Found existing installation: patch-ng 1.17.4
    Uninstalling patch-ng-1.17.4:
      Successfully uninstalled patch-ng-1.17.4
  Attempting uninstall: node-semver
    Found existing installation: node-semver 0.6.1
    Uninstalling node-semver-0.6.1:
      Successfully uninstalled node-semver-0.6.1
  Attempting uninstall: charset-normalizer
    Found existing installation: charset-normalizer 3.0.1
    Uninstalling charset-normalizer-3.0.1:
[... 1887 lines cut ...]
144446 INFO: Building EXE from EXE-00.toc completed successfully.
144476 INFO: checking COLLECT
144476 INFO: Building COLLECT because COLLECT-00.toc is non existent
144476 INFO: Building COLLECT COLLECT-00.toc
155806 INFO: Building COLLECT COLLECT-00.toc completed successfully.
```

python ./cura_inst/packaging/NSIS/create_windows_installer.py ./cura_inst/ ./dist/ cura.exe

```
Traceback (most recent call last):
  File "/home/fixed-command/cura/rapidiaCura5/temp/temp/./cura_inst/packaging/NSIS/create_windows_installer.py", line 82, in <module>
    build(args.dist_path)
  File "/home/fixed-command/cura/rapidiaCura5/temp/temp/./cura_inst/packaging/NSIS/create_windows_installer.py", line 72, in build
    subprocess.run(command)
  File "/home/fixed-command/.conan/data/cpython/3.10.4/_/_/package/41599e4c1f8b1b21e6c28e775ffeceed9a91f100/lib/python3.10/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/home/fixed-command/.conan/data/cpython/3.10.4/_/_/package/41599e4c1f8b1b21e6c28e775ffeceed9a91f100/lib/python3.10/subprocess.py", line 966, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/home/fixed-command/.conan/data/cpython/3.10.4/_/_/package/41599e4c1f8b1b21e6c28e775ffeceed9a91f100/lib/python3.10/subprocess.py", line 1842, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'makensis'
```

If there are any more questions I can answer to be more clear then please let me know.

jellespijker, 2023-03-08:
> `FileNotFoundError: [Errno 2] No such file or directory: 'makensis'`

It looks like you're missing a the build dependency **NSIS** make sure you add the `bin` dir with `makensis` to your `PATH` environment, once you have installed it. You can download NSIS from here: https://nsis.sourceforge.io/Main_Page

let me know if this solves it

thecampbells384, 2023-03-09:
That does indeed help, but now it cannot find script V2. I presume it probably won't be able to find script P4 either. Should these have been auto-generated somehow? 

```
Processing config: /etc/nsisconf.nsh
Can't open script "/V2"
```

More generally, when you were helping "yess" before in this issue https://github.com/Ultimaker/Cura/issues/13568,
you suggested that after installing from conan you should be able to run cura, and that this was a good sanity check that everything had worked to this point.

> @yess1987
>
> what happens if you remove the whole user data folder, profiles, data etc. from Conan. It looks like our custom generator
> aren't installed, or that you have old ones.

> first remove the following folders:
> C:\Users\Username\.conan and C:\.conan\

> Then run everything from the Powershell as a normal user:
```
pip install conan --force --upgrade
pip install sip
conan config install https://github.com/ultimaker/conan-config.git`
conan profile new default --detect
```
> now create a temporary folder somewhere cd into that and run
```
conan install cura/5.2.1@_/_ --build=missing --update -g VirtualPythonRunEnv -o cura:devtools=True -if cura_inst --json cura_inst/conan_install_info.json
```
> then activate the environment and run cura as a test
```
./cura_inst/Scripts/Activate.ps1
python ./cura_inst/cura_app.py
```
> if cura 5.2.1 starts up you can close it and run pyinstaller
```
pyinstaller ./cura_inst/Ultimaker-Cura.spec
```

In my case I cannot run the installed Cura after creating my own version in conan as it cannot find the plugins. Is this indicative of an error somewhere earlier in the build process, or is the conanfile just not configured to allow running cura in this way?

thecampbells384, 2023-03-22:
Hi, I am still struggling to get this to work. In the interests of rulling out system setup issues I created a Dockerfile based on my interpretation of your github actions.

```
FROM ubuntu:22.04

WORKDIR /Cura
SHELL ["/bin/bash", "-c"]

RUN apt update
RUN apt upgrade
RUN apt install python3 pip git -y
RUN python3 --version
RUN pip --version

RUN pip install conan==1.56.0 sip
RUN conan config set storage.download_cache="$HOME/.conan/conan_download_cache"

RUN apt remove libodbc2 libodbcinst2 unixodbc-common -y

RUN rm /var/cache/debconf/config.dat
RUN dpkg --configure -a
RUN apt-get install software-properties-common -y
RUN add-apt-repository ppa:ubuntu-toolchain-r/test -y
RUN apt update
RUN apt upgrade
RUN apt install build-essential checkinstall libegl-dev zlib1g-dev libssl-dev ninja-build autoconf libx11-dev libx11-xcb-dev libfontenc-dev libice-dev libsm-dev libxau-dev libxaw7-dev libxcomposite-dev libxcursor-dev libxdamage-dev libxdmcp-dev libxext-dev libxfixes-dev libxi-dev libxinerama-dev libxkbfile-dev libxmu-dev libxmuu-dev libxpm-dev libxrandr-dev libxrender-dev libxres-dev libxss-dev libxt-dev libxtst-dev libxv-dev libxvmc-dev libxxf86vm-dev xtrans-dev libxcb-render0-dev libxcb-render-util0-dev libxcb-xkb-dev libxcb-icccm4-dev libxcb-image0-dev libxcb-keysyms1-dev libxcb-randr0-dev libxcb-shape0-dev libxcb-sync-dev libxcb-xfixes0-dev libxcb-xinerama0-dev xkb-data libxcb-dri3-dev uuid-dev libxcb-util-dev libxkbcommon-x11-dev pkg-config nsis cmake -y

RUN apt install g++-12 gcc-12 -y
RUN update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-12 12
RUN update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-12 12

RUN conan profile new default --detect
RUN conan config install https://github.com/Ultimaker/conan-config.git

RUN git clone --branch 5.3 https://github.com/Ultimaker/Cura.git
WORKDIR /Cura/Cura
RUN conan create . cura/5.3@me/testing --build=missing --update

WORKDIR /installation
RUN conan install cura/5.3@me/testing --build=missing --update -if cura_inst -g VirtualPythonEnv --json "cura_inst/conan_install_info.json"

RUN conan install openssl/1.1.1l@_/_ --build=missing --update -o openssl:shared=True -g deploy
RUN cp ./openssl/lib/*.so* ./cura_inst/bin/ || true
RUN cp ./openssl/lib/*.dylib* ./cura_inst/bin/ || true  

RUN source cura_inst/bin/activate && pyinstaller cura_inst/UltiMaker-Cura.spec && python ./cura_inst/packaging/NSIS/create_windows_installer.py ./cura_inst/ ./dist/ cura.exe

CMD "bin/bash"
```

Alas, I still get the same error as before, that it can't open script V2. I note that when I run pyinstaller many files apear to be missing.
```
824 INFO: PyInstaller: 5.8.0
825 INFO: Python: 3.10.4
838 INFO: Platform: Linux-5.19.0-35-generic-x86_64-with-glibc2.35
845 INFO: UPX is not available.
924 WARNING: Unable to copy metadata for cura: The 'cura' distribution was not found and is required by the application
2963 INFO: Determining a mapping of distributions to packages...
13440 WARNING: Unable to determine requirements for cura: The 'cura' distribution was not found and is required by the application
13466 WARNING: Unable to copy metadata for UM: The 'UM' distribution was not found and is required by the application
14591 WARNING: Unable to determine requirements for UM: The 'UM' distribution was not found and is required by the application
14618 WARNING: Unable to copy metadata for serial: The 'serial' distribution was not found and is required by the application
15702 WARNING: Unable to determine requirements for serial: The 'serial' distribution was not found and is required by the application
409 WARNING: Failed to collect submodules for 'Charon.Service' because importing 'Charon.Service' raised: ModuleNotFoundError: No module named 'dbus'
16656 INFO: Packages required by Charon:
[]
16683 WARNING: Unable to copy metadata for sqlite3: The 'sqlite3' distribution was not found and is required by the application
17279 WARNING: Unable to determine requirements for sqlite3: The 'sqlite3' distribution was not found and is required by the application
19892 INFO: Packages required by trimesh:
['numpy', 'setuptools']
19959 WARNING: Unable to copy metadata for win32ctypes: The 'win32ctypes' distribution was not found and is required by the application
19961 WARNING: collect_data_files - skipping data collection for module 'win32ctypes' as it is not a package.
19963 WARNING: collect_dynamic_libs - skipping library collection for module 'win32ctypes' as it is not a package.
20211 WARNING: Unable to determine requirements for win32ctypes: The 'win32ctypes' distribution was not found and is required by the application
22617 WARNING: Unable to find package for requirement PyQt6-sip from package PyQt6.
22617 WARNING: Unable to find package for requirement PyQt6-Qt6 from package PyQt6.
22617 INFO: Packages required by PyQt6:
[]
22671 WARNING: Unable to copy metadata for PyQt6.QtNetwork: The 'PyQt6.QtNetwork' distribution was not found and is required by the application
22824 WARNING: collect_data_files - skipping data collection for module 'PyQt6.QtNetwork' as it is not a package.
23024 WARNING: collect_dynamic_libs - skipping library collection for module 'PyQt6.QtNetwork' as it is not a package.
23548 WARNING: Unable to determine requirements for PyQt6.QtNetwork: The 'PyQt6.QtNetwork' distribution was not found and is required by the application
[... cut ...]
24447 WARNING: Unable to determine requirements for PyQt6.sip: The 'PyQt6.sip' distribution was not found and is required by the application
24513 WARNING: Unable to copy metadata for stl: The 'stl' distribution was not found and is required by the application
25826 WARNING: Unable to determine requirements for stl: The 'stl' distribution was not found and is required by the application
```

Again, any help you can provide would be greatly appreciated. Anything you need from me just let it be known.

thecampbells384, 2023-04-11:
I determined that the main source of my problem was that the process of building a windows exe from linux does not work. In the end I made it work, including custom versions of fdm_materials and curaEngine with the following github action in place, and inspired by, your cura-installer action. The build process seems to be in quite the state of flux currently, but maybe someone will find this useful before things change too much for it to work as intended.

```
name: Cura Installer
run-name: ${{ inputs.cura_conan_version }} for ${{ inputs.platform }} by @${{ github.actor }}

on:
    workflow_call:
        inputs:
            platform:
                description: 'Selected Installer OS'
                default: 'ubuntu-20.04'
                required: true
                type: string
            os_name:
                description: 'OS Friendly Name'
                default: 'linux'
                required: true
                type: string
            cura_conan_version:
                description: 'Cura Conan Version'
                # must match format cura/[0-9].[0-9].[0-9]@whatever 
                # and the version numbers must match the version number in the first line of the conandata.yml file
                default: 'cura/5.0.1@rapidia/testing' 
                required: true
                type: string

env:
    CURA_CONAN_VERSION: ${{ inputs.cura_conan_version }}
    TOKEN_GITHUB: ${{ secrets.TOKEN_GITHUB }}

jobs:
    cura-installer-create:
        runs-on: ${{ inputs.platform }}
        steps:
            -   name: Checkout
                uses: actions/checkout@v3

            -   name: Checkout engine
                uses: actions/checkout@v3
                with:
                    repository: <my-cura-engine-repo>
                    token: ${{ secrets.TOKEN_GITHUB }}
                    path: curaEngine

            -   name: Checkout materials
                uses: actions/checkout@v3
                with:
                    repository: <my-materials-repo>
                    token: ${{ secrets.TOKEN_GITHUB }}
                    path: fdm_materials5

            -   name: Setup Python and pip
                uses: actions/setup-python@v4
                with:
                    python-version: '3.10.x'
                    cache: 'pip'
                    cache-dependency-path: .github/workflows/requirements-conan-package.txt

            -   name: Install Python requirements for runner
                run: pip install -r <my-cura-repo>/.github/workflows/requirements-conan-package.txt

            -   name: Create the default Conan profile
                run:  conan profile new default --detect

            -   name: Get Conan configuration
                if: ${{ inputs.conan_config == '' }}
                run: conan config install https://github.com/Ultimaker/conan-config.git

            -   name: create conan Engine package (powershell)
[... 62 lines cut ...]
                with:
                    name: ${{ steps.filename.outputs.INSTALLER_FILENAME }}-${{ steps.filename.outputs.INSTALLER_EXT }}
                    path: |
                        dist/*.tar.gz
                        dist/*.zip
                        dist/*.exe
                        dist/*.asc
                    retention-days: 5

Footer
© 2023 GitHub, Inc.
Footer navigation

    [Terms](https://docs.github.com/site-policy/github-terms/github-terms-of-service)
    [Privacy](https://docs.github.com/site-policy/privacy-policies/github-privacy-statement)
    [Security](https://github.com/security)
    [Status](https://www.githubstatus.com/)
    [Docs](https://docs.github.com/)
    [Contact GitHub](https://support.github.com/?tags=dotcom-footer)
    [Pricing](https://github.com/pricing)
    [API](https://docs.github.com/)
    [Training](https://services.github.com/)
    [Blog](https://github.blog/)
    [About](https://github.com/about)
    ```

closed 2023-04-11 by thecampbells384

jellespijker, 2023-04-28:
Great to hear that you fixed your problem