Ultimaker/Cura #14657: Error while slicing

Gotya0910, 2023-02-24:
### Application Version

5.3.0

### Platform

Windows 11

### Printer

Ender 3 V2

### Reproduction steps

I added a model and ran the slicing operation

### Actual results

Error while slicing

### Expected results

The model gets sliced

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

None

MariMakes, 2023-02-24:
Hey @Gotya0910,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

Gotya0910, 2023-03-02:
Hello @MariMakes 

Thank you for the warm welcome
I really wish I could provide more details, it so happened that after I had reported this issue, I went back to my Cura 5.3.0-beta 1
and I tried slicing again with the same parameters but with a different layer thickness and it worked perfectly.

I hope this helped

MariMakes, 2023-03-03:
Hey @Gotya0910,

Thanks for getting back to us 🙂 
Changing your layer height also changes other parameters, making your issue hard to troubleshoot. 😅 
I'm glad you got unstuck 🎉 

I'll close the issue, but if you have more info or run into another bug, feel free to start a new ticket.

closed 2023-03-03 by MariMakes