Ultimaker/Cura #14658: Normal supports not being generated

tangcla, 2023-02-24:
### Application Version

5.3.0-beta1

### Platform

Windows 11

### Printer

Creality CR-6 SE

### Reproduction steps

Used a different model - still no supports
Set support overhang angle to 10 degrees - still no supports
Toggled between normal and tree supports - tree supports are fine

### Actual results

No supports are generated in the preview

### Expected results

Normal supports should be created in the model as there are some serious overhangs (especially at 10 degree overhang angle).
Tree supports are created as expected, but normal supports are absent.
[image]
[bottom.zip](https://github.com/Ultimaker/Cura/files/10820037/bottom.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10820033/cura.log)

MariMakes, 2023-02-24:
Hey @tangcla,

Thanks for your report 👍
That doesn't look like it's going to print well 😮 

[image]

I'll bring it up to the team to see what they can do to improve before the stable.
Fingers crossed 🤞

MariMakes, 2023-02-24:
Hey @tangcla ,

Thanks again for bringing it up. 
We are super grateful for the report. 

We've added a ticket to with the intent to fix this as soon as possible. 
For internal reference CURA-10340

Thanks for the report, and apologies for the inconvenience.

MariMakes, 2023-02-24:
[image]

Update! It seems to be related to the experimental Conical Support. 
We'll be working on a solution, but at least you get unstuck.

teafusion, 2023-02-25:
I opened another ticket before finding this one, but have more notions on the issue aka using conical supports brim creation also breaks in conjunction when you disable prime tower while dual extrusion enabled, resulting the brim to become very narrow, possibly including only the second extruder two lines. My ticket https://github.com/Ultimaker/Cura/issues/14666

MariMakes, 2023-03-20:
Hey @teafusion,

Thanks to your report we were able to resolve the issue here: https://github.com/Ultimaker/CuraEngine/commit/34e126879add53fd15796d19a3e4e0a90740d38c

I can successfully slice your model in Cura 5.3 Stable, [you can download the stable here to try for yourself. ](https://github.com/Ultimaker/Cura/releases/tag/5.3.0)

[image]

Thanks again, thanks to your report we were able to narrow down what the problem was. 
I'll close this issue since there is a solution in place. 💪

closed 2023-03-20 by MariMakes