Ultimaker/Cura #14659: Slicing failure on 5.3.0-beta.2

ChrisMoller, 2023-02-24:
### Application Version

5.3.0-beta.2

### Platform

Fedora Linux Rel 33

### Printer

Ender 3 pro

### Reproduction steps

1. Started Cura
2. Opened an STL
3. The STL is rendered on the Prepare screen.

### Actual results

1. Slicing started
2. After a few seconds a window popped up announcing "Slicing failed with an unexpected error." and the Slicing... status bar just goes back and forth forever.

### Expected results

Slicing should have succeeded.
[frame.zip](https://github.com/Ultimaker/Cura/files/10820034/frame.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/10820039/cura.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Fails the same way with and without support generation.

MariMakes, 2023-02-24:
Hey @ChrisMoller,

Welcome to the Ultimaker Cura Github 🚀 

I took a look at your file. That's a risky printjob 😮 
I noticed that the file slices if you change the support placement. 
[image]

I also noticed that if you decrease the support angle to generate more sensible support, the first layer is missing, it might be related to adhesion set to None. 

I'll bring it up to the team to see what they can do to improve it. 
Thanks for your report 👍

casperlamboo, 2023-03-10:
Thank you once again for making us aware of the issue. We made some changes during the beta period regarding support generation. As a happy side effect this issue does now not occur any more in the latest 5.3 release.

[image]

closed 2023-03-10 by casperlamboo