Ultimaker/Cura #14662: [AppImage Modern] Desktop Icon is Missing on Ubuntu 22.10 (Gnome)

leoheck, 2023-02-24:
### Application Version

5.3.0-beta2

### Platform

Ubuntu 22.10

### Printer

n/a

### Reproduction steps

The desktop icon of AppImage (modern) is missing, still.

This was never working. Never! It was always broken like this.

This problem is the same as this one reported here:
https://github.com/FreeCAD/FreeCAD-Bundle/issues/155

### Actual results

The left icon was fixed by me in the previous version of Cura.

The right icon is the missing one form the newly installed Cura.
[image]
from

### Expected results

To fix this, the icon should be installed inside those folders:

```
./hicolor/128x128/apps/appimagekit_08778dffecd43a5e73c8b4396b0d8d2d_cura-icon.png
./hicolor/64x64/apps/appimagekit_08778dffecd43a5e73c8b4396b0d8d2d_cura-icon.png
./hicolor/256x256/apps/appimagekit_08778dffecd43a5e73c8b4396b0d8d2d_cura-icon.png
```

This is the icon fixed:

_Note: The old version of Cura was removed since the update of Cura was not able to remove the old version of it._

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

n/a

MariMakes, 2023-03-20:
Hey @leoheck,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

I'm a windows user myself so I've never encountered it. 
We have some Linux developers on the team, so they might have seen it. 

I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. Fingers crossed 🤞

leoheck, 2023-03-21:
Hi @MariMakes thanks for replying. Being a Linux user is a quite generic thing to say. It does not mean your colleagues will have the same issue I do since it depends on the Distribution/Window Manager.

I am using the Standard Ubuntu 22.10 which uses the Standard Gnome as the Window Manager. So if they are not using the same system, they may not see this problem.

But I hope you guys have time to tackle this problem because it is sad to use the Nice Ultimaker Cura without its icon/mark! Cheers!

leoheck, 2023-03-22:
Hi, Now I understand that I am using the [AppImage Launcher](https://github.com/TheAssassin/AppImageLauncher) that manages these files easily. 

I also found this [appimagelint](https://github.com/TheAssassin/appimagelint) by the same author. I am not sure if we have to follow it or how good this is, but at least it gives us a hint on where the issues may be. 

However, the appimagelint does not show any hint of where the issue might be. 

Is there any other way to check why the desktop Icons of Cura are not being added?

[image]

TheAssassin, 2023-03-22:
Cura is missing icons in the XDG icon theme directory `usr/share/icons`. As shown in https://github.com/Ultimaker/Cura/issues/14662#issue-1599151257, there used to be icons in those directories. Ideally, if you have an SVG, just put it into `usr/share/icons/hicolor/scalable/apps`, otherwise move your PNG(s) into the directory corresponding to its resolution (be aware only the typical resolutions are supported, e.g., 64x64, 128x128), e.g., `usr/share/icons/hicolor/64x64/apps/cura-icon.png`

leoheck, 2023-03-24:
I am linking here my old issue that can be closed when we fix this. 
https://github.com/Ultimaker/Cura/issues/12915

There is this routine in the `create_appimage.py`
```
def copy_metadata_files(dist_path, version):
    """
    Copy metadata files for the metadata of the AppImage.
    """
    copied_files = {
        os.path.join("..", "icons", "cura-icon_256x256.png"): "cura-icon.png",
        "cura.appdata.xml": "cura.appdata.xml",
        "AppRun": "AppRun"
    }
```
 
That copies the icons to an unknown place.

I am adding these guys here as well.
```
cura-icon.svg
cura-icon_64x64.png
cura-icon_128x128.png
```

I just have to figure out where the `copy_metadata_files` is copying the files to make sure they are going to end in the right place. 

@MariMakes can you check with your team if they can answer the above question?

MariMakes, 2023-03-28:
Hey @leoheck,

I brought your issue up with the team. 
When we discussed it we saw that you already made a PR, super cool 😍 👑 

Our top-shelf Linux Developer is going to take a look.
Feel free to ping me if you are stuck 👍

leoheck, 2023-03-28:
Awesome, I appreciate it.

closed 2023-04-28 by jellespijker

cyberlab-dev, 2023-04-28:
Not working..

commit e8ef90da (2023-04-28): Merge pull request #15015 from leoheck/fix-appimage-icons

Fix AppImage icons on some Linux systems

Fixes #12915 
Fixes #14662

Implements CURA-10550