Ultimaker/Cura #14666: Conical support breaks support and brim in conjunction with disabling prime tower

teafusion, 2023-02-24:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Ultimaker 3

### Reproduction steps

Load an STL file to Cura, one that needs support, then assign the object to extruder 2 with AA 0,8mm nozzle, while extruder 1 has AA 0,4mm nozzle, both extruders needs to be active. Reset profile to Cura's default profile (fast 0.2) if not already and then enable support. Then enable conical support from experimental. Next disable dual extrusion prime tower.

### Actual results

Support is gone by enabling conical support and brim gets very narrow by disabling dual extrusion prime tower when support and conical support are enabled. Seems that conical support kills the supports and disabling prime tower makes the brim very narrow. In both cases support/conical support is the key factor to broken features.

### Expected results

Support should exist and be conical and the brim should be normal as defined.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[UM3_Simple_sample__LOG.zip](https://github.com/Ultimaker/Cura/files/10828978/UM3_Simple_sample__LOG.zip)

closed 2023-03-20 by MariMakes

teafusion, 2023-03-21:
Thank you @MariMakes, no problem at all.

Except the brim is not 7.0mm or 9 lines wide. It seems that disabling support the brim is sliced to what it's set.

MariMakes, 2023-03-21:
Hey @teafusion,

What you mention is a known issue. 
In 5.3 we introduced brims that can be printed in multiple materials. 
Some people encounter small brims if they have not defined the Build Plate Adhesion Extruder. 
If you define that, it shows up correctly like this;
[image]

Hope this helps 👍

teafusion, 2023-03-21:
Aaw, alright. Thank you for clarification! In dual extrusion, with two different materials and two different nozzles I didn't much care about setting the brim extruder before, because it would always default to the first extruder, which in my case has always had 0,4mm nozzle, which creates easier to remove brims in all cases, whereas in my case 2nd extruder and 0,8mm produces such strong brim that it's very difficult to remove nicely, unless you increase brim's distance from the object to something like 0.5-0.6mm, which I've used when printing with only one extruder and 0,8mm nozzle.

Hope though that this unexpected mini brim could be chased out of the Cura, so that you could always trust the settings you put in yourself.