Ultimaker/Cura #14669: Object slice fails in 5.3.0-B1 - works fine in 5.2.1

Zotter, 2023-02-25:
### Application Version

5.3.0-beta.1

### Platform

Fedora 37

### Printer

Anet A8

### Reproduction steps

Downloaded and unzipped this object:  https://www.thingiverse.com/thing:3104762

Attempted to slice, throws: "Slicing failed with an unexpected error." - no more information offered

Loaded into 5.2.1 and sliced with same settings and no problem.

### Actual results

[project-file.zip](https://github.com/Ultimaker/Cura/files/10830058/project-file.zip)

### Expected results

[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10830061/cura.log)

WillCreate1, 2023-02-25:
Had the same thing happen tried three times, then changed supports from Normal to Tree and it sliced it. Not sure if its something to do with the supports

MariMakes, 2023-03-20:
Hey @Zotter,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

The new Cura 5.3 stable contains the fix for this for a number of slicing crashes. 
I was able to slice your project file in 5.3.0 stable. Could you test if that's the case for your model as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0

[image]

If that's not it, can you doublecheck if you shared the correct file and let us know. 
Than I'll take another look 👍

Zotter, 2023-06-22:
Sorry so late and Thanks! Yes, can confirm the new version fixed it nicely.

closed 2023-06-22 by Zotter