Ultimaker/Cura #14671: Infill input issue

rdfry, 2023-02-25:
### Application Version

5.0 5.1 5.2 5.2.1 5.2.2

### Platform

Windows 11

### Printer

Ender3 Pro

### Reproduction steps

input percentage in infill column

### Actual results

get knocked out of infill column happens until final percentage is input. this was a known problem in 5.0 .  happens for me in 5.1 also.

### Expected results

entire percentage should be able to be input at one time.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no files to add

rdfry, 2023-03-07:
5.2.2 as well as 5.2.3 beta

MariMakes, 2023-03-20:
Hey @rdfry,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

I recognize your report. 
This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/12949
We have a ticket on our backlog with the intent to improve this behavior.

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12949

closed 2023-03-20 by MariMakes