Ultimaker/Cura #14672: Multiply objects causes covering objects

quiiiebong, 2023-02-25:
### Application Version

5.3.0-beta.2

### Platform

Windows 11

### Printer

Anycubic i3

### Reproduction steps

Load object. right click, multiply object, amunt:1

### Actual results

the new object is not seperated but covers the source object

### Expected results

Objects shall be separated. worked well with cura 5.2.2

scrennshot of wrong behaviour
[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Object: https://www.thingiverse.com/thing:950452/files -> wind wheel mount

MariMakes, 2023-03-07:
Hey @quiiiebong,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

I was able to reproduce your behavior.
But they did not overlap once I had the "Ensuer Models are kept Apart" checked in the preferences:
[image]

To me, it doesn't seem like a bug in Ultimaker Cura anymore, since there is a solution in place. 
Would you still like to keep this issue open or do you agree that we can close it?

quiiiebong, 2023-03-07:
Hello @MariMakes !

First - Thank You! - for having a look on the issue.

I confirm that with setting that switch the objects are now kept apart. However, in 5.2.2 it worked in default settings without activating this option. 5.3 does not deliver a printable result with the default installation settings. This should be the case.

Also, the placing is much worse than in 5.2.2. I opened the same set of objects in 5.2.2 and 5.3. While the result of 5.2.2 is usable without manual action, the result of 5.3 is not (the switch suggested by you enabled...)

  
Cura 5.2.2 (expected behaviour):
[image]

Cura 5.3
[image]

The behaviour changed to the worse, so I would still see it as a bug.

MariMakes, 2023-03-20:
Hey @quiiiebong,

Sorry, it took us a while to get back to you 😞 
We've seen more reports of this. 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/14112
We have a ticket on our backlog with the intent to improve this behavior.

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/14112
Thanks for your report 👍

closed 2023-03-20 by MariMakes