Ultimaker/Cura #14674: Possibility to add/remove extensions to/from Cura right-click menu

teafusion, 2023-02-25:
### Is your feature request related to a problem?

I'm often frustrated about having to go to the extensions top menu bar for my favorite and most frequently used extensions. It would be nice, if you could decisively add/remove extensions to/from the right-click menu in either extensions own settings or from general Cura settings, so that you could selectively customize the right-click with some extension you prefer for more immediate application. This would be especially handy when you have more than one printable at the time and you need a little different treatment on each other, then going back and forth to the top menu bar for the needed extensions, while selecting and unselecting objects one at the time causes unnecessary friction the workflow.

### Describe the solution you'd like

Possibility to add/remove extensions to the right-click menu, either by each extensions own settings or from Cura's global settings.

### Describe alternatives you've considered

None.

### Affected users and/or printers

All users would benefit who use extensions a lot, or should you call them power-users then.

### Additional information & file uploads

I suggested this already to Nallath for a feature to add to Auto-orientation extension, but then thought why not present the idea to the Cura community in hopes to make it a global Cura feature. Currently only Mesh Tools has been added to the right-click, which is handy and I use it frequently, although Mesh Tools has so many commands that may apply to all imported printables at once, so I often use it also from the top bar menu alternatively, but I'd say 60/40 advantage for the right-click use anyway.

fieldOfView, 2023-02-25:
I think it is up to the individual plugin creators to implement it. For most plugins it does not make sense to have entries in the context-menu. The MeshTools implementation is here:
https://github.com/fieldOfView/Cura-MeshTools/blob/release/MeshTools.py#L114-L141

teafusion, 2023-02-25:
Thank you for the input. Agree there are extensions that don't have real use or would benefit anything much from being in the right-click menu, but there are still many, that would be handy over there. Of the extensions that I have installed, some of them in more frequent than others, yet all that I would at least consider adding to my right-click if it was an enable/disable universal option:

- Parts for calibration (would be nice if the parts were also created at place of the mouse point)
- Cura Blender (would probably use more from the right-click)
- Mesh Tools (already there, hence not an issue, would consider removing even, if the right-click menu got too crowded from other, more frequently used extensions)
- Auto-orientation (definitely would add and keep there)
- Thingibrowser (why not, would probably use more often, rather than use a browser, especially if there was a process to download and open the object at mouse point)
- Universal settings (would probably use more often, it's been mostly just hanging out there)

Mesh Tools is kind of a strange oddity now, when it's the only one in right-click of those that I have installed, so it seems almost like a policy breaker in that sense. Right click menu use is not a norm for example in Windows desktop and there's definitely unified policy for that, but some programs that offer the opportunity to be added to "context menu" are nice to have there, optionally.

If nothing else, should there be a policy for Cura, that the extension makers would actively consider at least possibility to add theirs to the right-click, if it wasn't made available as a global choice for the user, of course you'd need to have expert or all options showing to be able to do this. All in all this is mainly a matter of user experience or usability and options to modify it with at least some relative ease, without having to make your own fork (which I'm pretty impaired to do, being mainly a Cura power user, rather than a coder or software engineer).

5axes, 2023-02-26:
Make sens only if you need to select the part before to apply a function. So mesh-tool, auto-orientation yes perhap's NameIt and even with this one I'm not sure. But one enhancement on the Cura side would be the possibility to define a key-board shortcut natively on the extension menu.