Ultimaker/Cura #14679: 'dict object' has no attribute 'polygon'

ckvsoft, 2023-02-25:
### Application Version

5.3 Beta Appimage

### Platform

Ubuntu 20.04

### Printer

everyone 

### Reproduction steps

Slice, upload to printer and use exclude objects map

### Actual results

"BED_MESH_CALIBRATE  'dict object' has no attribute 'polygon'"

### Expected results

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

With 5.3 Beta 1 I have the problem that exclude objects no longer works. Is this a bug or was it changed on purpose?
With version 5.2 everything works with the 5.3 Beta1 I get the error in klipper or also via Octoprint: dict object' has no attribute 'polygon'
Unfortunately I can't describe it any better.

ckvsoft, 2023-02-27:
Ok
It is a Typo in Ultimaker/CuraEngine LayerPlan.cpp

martinziener, 2023-03-07:
What's new is in the gcode this gets added:
**EXCLUDE_OBJECT_DEFINE NAME=NOMESH**

in klipper the printer.exclude_object.objects | map(attribute='polygon') doesn't like that

gillesguillemin, 2023-03-08:
This causes some major issues with Klipper indeed (like breaking Adaptive Meshing).

ckvsoft, 2023-03-08:
Its a Typo in CuraEngine
https://github.com/Ultimaker/CuraEngine/pull/1855

ckvsoft, 2023-03-08:
A workaround is to use the search and replace post processing script

search NOMESH
replace NONMESH

martinziener, 2023-03-08:
no, it's not (just) a spelling error
!! Error evaluating 'gcode_macro Adaptive_MESH:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'polygon'

martinziener, 2023-03-08:
I just found out that it's not cura who generates the EXCLUDE_OBJECT_DEFINE-gcode but that klipper adds it to the gcode at upload.

But why does klipper suddenly see two objects where before it only saw one?

ckvsoft, 2023-03-08:
For me, renaming works in the GCode that Cura generates. This must of course be done before sending it to the printer.
Thus will
EXCLUDE_OBJECT_DEFINE NAME=NOMESH not created at all.
In any case, an Exclude Object and an Adaptive mesh work for me.

Changing the GCode in Mainsail / Fluid does nothing, it has to be changed in the original gcode

martinziener, 2023-03-08:
if i save the gcode that cura generates to a file, there is no mentioning of a EXCLUDE_OBJECT_DEFINE.
But uploaded it gets added

ckvsoft, 2023-03-08:
> if i save the gcode that cura generates to a file, there is no mentioning of a EXCLUDE_OBJECT_DEFINE. But uploaded it gets added: preprocess-cancellation. v0.2.0

Yes. In the saved gcode, change NOMESH to NONMESH
Then it works. You can let Cura do it right away.

martinziener, 2023-03-08:
Ah ok, I didn't know that.
How can I let Cura write the EXCLUDE_OBJECT_DEFINE gcode lines?

ckvsoft, 2023-03-08:
> Ah ok, I didn't know that. How can I let Cura write the EXCLUDE_OBJECT_DEFINE gcode lines?

No, just replace NOMESH

Cura -> Menu ->  Extensions -> Post Processing -> Modify GCode

Then Add a Script -> Search and Replace
Search is NOMESH
Replace is NONMESH

thats all

martinziener, 2023-03-08:
Weird, but that works.
Thank!

henryabra, 2023-03-10:
@ckvsoft Worked for me! Thanks

MariMakes, 2023-03-20:
Hey All,

Sorry, it took us a while to get back to you 😞 
Thanks for the report!

I'll bring it up to the team to see what they can do to improve it.
It seems to be just a typo. Thanks for the investigation and the workaround @ckvsoft ❤️ 

Fingers crossed that we can share a fix soon🤞

ckvsoft, 2023-03-24:
fixed in CuraEngine #1855

closed 2023-03-24 by ckvsoft

Extrusions3D, 2023-04-15:
SO DOES ANYONE KNOW WHY THE CURA 5.3.0 has been out for a minute but they have not released 5.3.1 with the typo fixed so we can continue using KAMP without modify g code?

MariMakes, 2023-05-04:
Hey @Extrusions3D,

Cura 5.3.1 should be available. 🎉 
You can download it here https://github.com/Ultimaker/Cura/releases/tag/5.3.1

We had a delay because implementing the updated translations seemed to be more work than we anticipated. 😞 
But that's resolved 💪

Johnn201, 2023-07-05:
superslicer has this same issue Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': UndefinedError: 'dict object' has no attribute 'polygon'

ckvsoft, 2023-07-05:
That's already fixed in Cura. Superslicer does not depend on Cura.

hypersmc, 2023-07-26:
> That's already fixed in Cura. Superslicer does not depend on Cura.

The fix is NOT just a typo... ALL of you have ignore the 1 user who acually found the issue. 

[martinziener](https://github.com/martinziener) found the issue and all of you never care about the actual fix. 

Remove EXCLUDE_OBJECT_DEFINE NAME=NOMESH or if you renamed it EXCLUDE_OBJECT_DEFINE NAME=NONMESH
does the job.

MariMakes, 2023-07-27:
Hey @hypersmc,

We see that there is a rise in 3D printers that are running Klipper firmware. As UltiMaker employees we focus on the motion controller that our UltiMaker printers use. So we are less exposed to the finer details of Klipper. But we are interested and want Cura and Klipper to work smoothly together.

We are collecting input on how we can make Cura better for Klipper printers. If you have some feedback, we would love to hear about it here:  https://github.com/Ultimaker/Cura/discussions/16012

d3dorsett, 2023-12-05:
Still does not appear to be fixed in Cura 5.6.0. KAMP and Gcode Preview will not work in Klipper.

d3dorsett, 2023-12-05:
FYI: KAMP and GCode Preview work correctly for me in Cura 5.6.0, with fluidd (klipper)

tipitorsten, 2024-03-31:
Cura 5.6. have incorect EXCLUDE_OBJECT_DEFINE NAME for me. Kamps doesent work with cura on klipper.  orca slicer works