Ultimaker/Cura #14681: Strange layers

IngwarSa, 2023-02-25:
### Application Version

5.3.0-beta2

### Platform

Windows 10

### Printer

Voron 2.4

### Reproduction steps

Wall thickness 2.5mm
Layer width 0.5mm

### Actual results

Why are the layers on the corners of this shape?

### Expected results

Sorry if I wrote to the wrong address.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
[image]
[image]
[V250_1.zip](https://github.com/Ultimaker/Cura/files/10831952/V250_1.zip)

MariMakes, 2023-03-07:
Hey @IngwarSa,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Ah, this seems related to the Variable Linewidth Engine in Cura 5.x .

​​​​​​​There are settings that you can change to improve this behavior.
You can read more about them here: 
https://support.ultimaker.com/hc/en-us/articles/4792077687068

To me, it doesn't seem like a bug in Ultimaker Cura, because this behavior is setting dependend. 
Would you still like to keep this issue open or do you agree that we can close it?

MariMakes, 2023-03-07:
[image]

I think the setting you are looking for is the Wall Transitioning Threshold Angle. 

I'll close the issue! Good luck with your printjob and let us know if you run into anything else 👍

closed 2023-03-07 by MariMakes