Ultimaker/Cura #14688: unable to slice

sirdirk22, 2023-02-26:
### Application Version

5.3.0-alpha+cura_9879_19

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

loaded stl from thingiverse
attempted to slice
recieved error
restarted cura
[image]

tried again with same result
stl can be found here https://www.thingiverse.com/thing:5159220

### Actual results

error. unable to slice

### Expected results

the file should be sliced and produce a gcode

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10832920/cura.log)
[Unable to Slice.zip](https://github.com/Ultimaker/Cura/files/10832927/Unable.to.Slice.zip)

sirdirk22, 2023-03-21:
> Hey @sirdirk22 <https://github.com/sirdirk22>,
>
> Welcome to the UltiMaker Cura Github 🚀
> Sorry, it took us a while to get back to you 😞
>
> It seems like your model is smaller than your Initial Layer Height.
> If you reduce your Initial Layer Height to 0.19 instead of 0.24 you are
> able to slice this model:
> [image: image]
> <https://user-images.githubusercontent.com/40423138/226639724-47451677-bf26-4c05-afdf-e04de99b69ac.png>
>
> To me, it doesn't seem like a bug in Ultimaker Cura anymore.
> Would you still like to keep this issue open or do you agree that we can
> close it?

oh man... how did i miss that? it was weird because it would slice just
fine in cura 5.2... please close this ticket :)  and thank you for your
time and attention 👍

MariMakes, 2023-03-23:
Glad to hear. 
Good luck with your door lock shim 🙂

closed 2023-03-23 by MariMakes