Ultimaker/Cura #14693: Cannot build from source (Arch)

bullestock, 2023-02-26:
### Application Version

master

### Platform

Arch Linux (kernel 6.1.12-arch1-1)

### Printer

N/A

### Reproduction steps

First I tried the steps listed at https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source, and that failed at the `conan profile new` step. Another issue led me to reinstall Conan using `conan profile new`, and that worked until the `conan install` step which fails with the error shown below.

### Actual results

`ERROR: Error loading conanfile at '/home/torsten/sw/Cura/conanfile.py': Version range '>=1.0.0' from requirement 'translationextractor/[>=1.0.0]@ultimaker/stable' required by 'py_require' could not be resolved in local cache`

See attachment for full Conan output.

### Expected results

I expected the build step to succeed.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[conan-log.txt](https://github.com/Ultimaker/Cura/files/10833814/conan-log.txt)

jellespijker, 2023-03-08:
Hi @bullestock a lot has changed with respect to the translations workflow, I believe this should now be fixed. Can you try it again. Ensure that you do a `git pull` and that you have the latest conan configuration installed `conan config install https://github.com/Ultimaker/conan-config.git`

Can you let me know if this solves your issue

bullestock, 2023-03-11:
I now get
```
xorg/system: WARN: Do not know how to install 'xorg' for endeavouros.
ERROR: xorg/system: Error in package_info() method, line 99
	self._fill_cppinfo_from_pkgconfig(name)
while calling '_fill_cppinfo_from_pkgconfig', line 25
	if not pkg_config.provides:
	ConanException: pkg-config command ['pkg-config', '--print-provides', 'xtrans', '--print-errors'] failed with error: Command 'pkg-config --print-provides xtrans --print-errors' returned non-zero exit status 1.
Package xtrans was not found in the pkg-config search path.
Perhaps you should add the directory containing `xtrans.pc'
to the PKG_CONFIG_PATH environment variable
Package 'xtrans', required by 'virtual:world', not found
```
This is still with Conan 1.56.0, as the default version still fails during the`profile new` step.

jellespijker, 2023-03-11:
which OS and architecture do you use and which system package manager?

bullestock, 2023-03-11:
@jellespijker `uname -a`
```
Linux orisha 6.2.2-arch2-1 #1 SMP PREEMPT_DYNAMIC Wed, 08 Mar 2023 04:07:29 +0000 x86_64 GNU/Linux
```
`lsb_release -a`
```
LSB Version:	n/a
Distributor ID:	EndeavourOS
Description:	EndeavourOS Linux
Release:	rolling
Codename:	rolling
```
CPU is AMD Ryzen 7 PRO 4750U. I use `yay` as my package manager

jellespijker, 2023-09-23:
please update conan to at least version 1.60.2 but less then 2 and make sure you follow the instructions on the Wiki, which mention that xtrans might need to be installed