Ultimaker/Cura #14695: HAVE TO RELOAD PRINT WITH F5, EVERY PRINT HAS TO BE DONT THIS WAY

B1gRon, 2023-02-26:
### Application Version

5.3.0 B2

### Platform

WIN 10

### Printer

ANYCUBIC VYPER

### Reproduction steps

Every time i load a file, its 300% bigger and fills the screen, and have to reload with F5 every time, this has been happening for a while some wear around late Version 4

### Actual results

after hitting F5, the model is reloaded at proper size and prints normely

### Expected results

the above results are the expected outcome

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no other info required

fieldOfView, 2023-02-26:
Do you have the MeshTools plugin installed? If so, go to Extensions -> Mesh Tools -> Mesh Tools settings... and make sure the `Unit for files that don't specify a unit` option is set to `Millimeter (default)`

fieldOfView, 2023-02-26:
If the above does not solve your issue, please provide the logs as requested.

B1gRon, 2023-02-26:
That seem to of been the issue, you guys Rock!! thanks

B1gRon, 2023-02-26:
Problem solved by FieldOfViep, wrong measurement setting by default in mesh tools

closed 2023-02-26 by B1gRon