Ultimaker/Cura #14698: experiencing issue with quality profiles for "fine"

Drayson80, 2023-02-26:
### Application Version

5.1.1, 5.2.1, 5.2.2, 5.3beta

### Platform

Win11

### Printer

UMO

### Reproduction steps

update from 5.1.0 to the next releases - latest 5.3beta too
open up Cura, try to change the quality setting, 
recognized that Normal and Fine quality profiles are both 0.15 (!?!?) instead of 0.15 and 0.1... 
Bug or related to the discontinued UMO? **In 5.1.0 all was fine...**

### Actual results

Normal quality profile is 0.15 => ok
Fine quality profile is also 0.15 (instead of 0.1) => nok!
Extra Fine profile is still 0.06 => ok

### Expected results

Fine quality profile is 0.1

so the same state as in 5.1.0

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

hava not made any changes at my profiles since ages, as Fine is my go-to profile always on the run. never experienced any issues during migration to newer versions.

MariMakes, 2023-03-21:
Hey @Drayson80,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I've had multiple Ultimaker employees look at your issue but we cannot reproduce the behavior you are reporting. 

Do you mind sharing some screenshots to help us understand your problem? 
Or perhaps a screen recording?

Drayson80, 2023-04-05:
Hi Mari,
decided to reset my printer profile and re-did it from scratch incl. a cleaned cura 5.3 install without any migrated profiles.
Nevertheless, I have still this issue when including one of my old profiles. see attached screenshot.
[image]

MariMakes, 2023-04-06:
Hey @Drayson80,

Thanks for getting back to us. 
That printer doesn't look like one we ship with Cura 🤔 
It doesn't look like an Ultimaker Original or a UMO+ DXU Dual.
How did you add this printer to Cura? 

In this case we can help you better if you share you configuration folder. 
You can share your configuration folder by:
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version
	
You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

Drayson80, 2023-04-07:
Hey @MariMakes ,
this is a heavily modified Ultimaker Original back from 2012, with Dual Extruder kit, but arranged to line up at X-axis, with a self-made heated bed. I renamed the printer to UMO Dual Extrusion CE (my initials). The definition file has been automatically migrated by cura since ages. 
I did some more investigation. in cura 4.13.0 everything works as it should, after instalation of 5.x, I get this strange issue.
It seems that there was a corrupt log. cfg or json file which made the trouble. 
[image]

I re-did the whole customization based on actual ultimaker_original definitions and now it seems to work again.
[4.13.zip](https://github.com/Ultimaker/Cura/files/11177408/4.13.zip)

I have no idea if this is related to the migration from 4.13.0 to 5.2x or 5.3 also for other users...
Attached my 4.13 config without materials and plugins. I use two profiles of the same printer - one for single nozzle prints to avoid heatup of the unused nozzle, one for dual prints. further there are 2 expermental profiles with a UM2 head and nozzle switch included which are not used/working at all. pls. ignor them...

b.r.
christian

MariMakes, 2023-05-03:
Hey @Drayson80,

Thanks for the update👍
I'm glad to hear you were able to resolve your issue. 😊 

We had to introduce an upgrade script in Cura 5.3 because we've done a big cleanup in the Ultimaker Profiles to make them future-proof. It could be that that change introduced some trouble for your custom set-up. 
This is the change I'm talking about https://github.com/Ultimaker/Cura/pull/14423

Since your issue has been resolved, I'll be closing this issue.
Thanks for the report!

closed 2023-05-03 by MariMakes