Ultimaker/Cura #14699: Генерация мостов Bridge generation

IngwarSa, 2023-02-26:
### Application Version

5.3.0-beta2

### Platform

Windows 10

### Printer

Voron 2.4

### Reproduction steps

Sometimes bridges are cut incorrectly

### Actual results

3 layers in a row in one direction

### Expected results

The direction in the layers should change

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[brige.zip](https://github.com/Ultimaker/Cura/files/10834767/brige.zip)

IngwarSa, 2023-03-21:
Thank you for your consideration. It's just that there were fewer settings in early versions, but most models were prepared correctly with default settings. And now there are a lot of settings, but you need to constantly adjust something to get a normal result ...(

MariMakes, 2023-03-23:
Hey @IngwarSa,

We try to improve Cura with every release.
That often means adding more settings or tuning existing settings, based on research. 

If you want to learn more about the settings we've had a community member write a helpful plug-in that adds more elaborate descriptions to the plug-ins. You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/Ghostkeeper/SettingsGuide2

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
I'll be closing this issue. I hope you understand. 
Happy printing!