Ultimaker/Cura #14700: Break up supports in chunks - doesn't break up anything

tlhintoq, 2023-02-27:
### Application Version

5.3.0-Alpha or beta

### Platform

Windows10x64, 21H2

### Printer

All - Not printer specific

### Reproduction steps

Put in a model.  
Support it
Choose to break up supports.

### Actual results

Supports are clearly not getting broken up into chunks.  One massive wall of support as shown.

### Expected results

There should be chunks of supports as the name implies.  Broken into segments (aka 'chunks') that can be removed indivually.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

See attachements and screen shots. No crash so no log attached.
[image]
[ChunksNotChunking.zip](https://github.com/Ultimaker/Cura/files/10835204/ChunksNotChunking.zip)

MariMakes, 2023-04-03:
Hey @tlhintoq,

Sorry, it took us a while to get back to you 😞 

Break support up on chunks is not a setting I use often so I had to ask around.

I've been told that the setting breaks up long zig zag support. See image without and with (see red arrows) attached to this thread. It indicates after how many zigzag infill lines it will remove a connecting zigzag line. It works best if your support wall is 0. 

[image]
[image]

It still works as intended in Cura 5.3.
Apologies for any inconvenience caused by our delayed response. 
I'll be closing this issue. I hope you understand.

closed 2023-04-03 by MariMakes

MariMakes, 2023-04-04:
Hey @tlhintoq,

Currently the tooltip reads: 
_"Skip some support line connections to make the support structure easier to break away. This setting is applicable to the Zig Zag support infill pattern.",_

How would you improve it?

tlhintoq, 2023-04-04:
You're right it does.  I stand corrected.