Ultimaker/Cura #14703: [AppImage Modern] Missing Workspace

tophers, 2023-02-27:
### Application Version

5.3.0-beta.2

### Platform

Linux

### Printer

Custom

### Reproduction steps

1.) Open Modern AppImage (Curent Beta or Stable)
2) I can open a file, change slicer settings, even slice a file and export/save it
3.) no workspace displayed

### Actual results

Workspace not Displayed

### Expected results

To be able to see/Rotate object visually

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/10836508/cura.log)

tophers, 2023-03-06:
I've since found this happens on any version of Cura regardless of install method, Flatpak, Native (apt, older version), appImage Modern or not. I've tried with a new user, same effect. What would cause just the workspace to not appear?

tophers, 2023-03-08:
> Could it be that the QML engine is not "seeing" the size of the window (and thusly sets some widths and heights to 0px)?
How would I verify or confirm this?

Joseja8, 2023-03-14:
This is happening to me too. Let me know if I can help.

tophers, 2023-03-18:
Console message FWIW:

$ ./UltiMaker-Cura-5.3.0-beta.2-linux-modern.AppImage 
PyInstaller/loader/pyimod02_importers.py:352: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_UltiMaeaLbfP/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMaeaLbfP/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
file:///tmp/.mount_UltiMaeaLbfP/share/cura/resources/qml/PrintSetupSelector/Recommended/RecommendedPrintSetup.qml:96:17: QML QQuickItem: Binding loop detected for property "height"
file:///tmp/.mount_UltiMaeaLbfP/share/cura/resources/qml/PrintSetupSelector/Recommended/RecommendedPrintSetup.qml:96:17: QML QQuickItem: Binding loop detected for property "height"

tophers, 2023-03-22:
> I cannot spot what is going wrong but I'll bring it up with the team. Maybe they can see what is happening or what information we need for troubleshooting.
> 
> Fingers crossed crossed_fingers

@MariMakes  Thank you. Willing to provide logs/test anything.

MariMakes, 2023-04-06:
Hey @tophers,

We have a nightly build available that might fix what you are encountering.
There is a slime chance, but it's still a chance that this fixes your issue. 

Can you give it a go and let us know if this resolves your issue?
[You can download the build here.](https://github.com/Ultimaker/Cura/discussions/15098)

tophers, 2023-04-06:
Still unable to see most windows within the application.  I honestly figure this is less of an application bug and more of something on my system -- but I don't know what or how those are generated or by what libraries may be an issue - or how to even setup an environment to troubleshoot. 

[image]
[image]
[image]

tophers, 2023-09-07:
Closing this as I'm no longer using Cura.

jellespijker, 2023-09-07:
@tophers since this seem to affect more people than just you, I'm reopening it because it is still an issue.
For the 5.5.0 release I'm using a different tool to collect the libraries I'm hoping if you would be willing to test if this AppImage runs for you: [UltiMaker-Cura-5.5.0-alpha+cura_10951_2bc307-linux-X64-AppImage](https://github.com/Ultimaker/Cura/suites/15883868038/artifacts/907218000) 

Can you also test this @Joseja8 ?

The AppImage is build on the PR: https://github.com/Ultimaker/Cura/pull/16607

MariMakes, 2023-10-19:
👋 Quick update on our side. 

This is resolved in the 5.5 Beta release 🎉 , you can download the Cura version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)
I'll close this issue since it's resolved. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2023-10-19 by MariMakes

quickishfm, 2024-07-08:
Hi, I'm facing this same issue on Cura 5.7.2. Running KDE Plasma 6.2.1 on Arch Linux (Appimage version - the one on the AUR does not build due to an orphaned dependency)

I'm using Nvidia graphics, in case that helps. I'm willing to provide logs and debug information to help get this fixed.

jellespijker, 2024-07-18:
what happens when you start Cura with `QT_QPA_PLATFORM=xcb ./UltiMaker bladibla.AppImage` 

written on my phone, replace with actual AppImage filename

quickishfm, 2024-07-19:
Thanks @jellespijker that makes it work!

I cannot remember if I wrote about the issue on X11 or Wayland (as during that time I was swapping back and forth due to some X11 KDE bugs) but I notice now that Cura launches fine on X11, but not on Wayland.
Using `QT_QPA_PLATFORM=xcb` fixes it on Wayland though.

jellespijker, 2024-07-19:
@HellAholic do we have a ticket for Wayland support on the backlog already?