Ultimaker/Cura #14704: z starts in 5.2.1 or maybe 5 plus are INSANE in comparison to 4.13.1

Hiaber, 2023-02-27:
### Application Version

5.2.1

### Platform

windows 10

### Printer

anycubic

### Reproduction steps

z starts on version 5.2.1 ( which is the latest i was running) until i upgraded today to the 5.3 beta - are toooo many - it messes up my prints, i've tried everything!  ive set the parameters the same as my version 4.13.1 and it slices it with one z seam - 5+ has hundreds of starts!  please see screenshots for details - note all the white line "starts" in each.  I have had to revert back to 4.13.1 for all my important prints because all the z starts make horrible prints!!!

### Actual results

i've tried changing almost every parameter to get rid of this problem with no luck

### Expected results

see screenshots uploaded - contact me if clarification is needed

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
[cura drive belt slice comparison betwee 4_13_1 and 5 plus.zip](https://github.com/Ultimaker/Cura/files/10837406/cura.drive.belt.slice.comparison.betwee.4_13_1.and.5.plus.zip)

GregValiant, 2023-02-27:
Thanks for the report and the project files.
When software is updated there is often a learning curve as we get to know the nuances of the changes that were made.
In this case the Variable Line Width of the 5.x versions did away with the "Compensate Wall Overlaps" which you used to good effect in the 4.13.1 project.  That setting has been replaced by a couple of other settings.  In this case, setting the "Wall Transitioning Threshold Angle" to 2.0 will give you a much cleaner print.

This is sliced in 5.2.1.  You can see that most of the starts are gone (5 remain).
[image]

The 5.2.1 slice is 5 minutes faster because it doesn't have all of this extra motion and lines (in yellow) that the 4.13.1 slice has.
[image]

I'm inclined to remove the bug label and close this as it appears to be a settings issue.  I'll hold off so you can respond if you think that would be premature.

Hiaber, 2023-02-27:
Thanks for your quick reply, I will try those settings and I agree that 
it's not a bug, feel free to close the issue

-John