Ultimaker/Cura #14707: Prime Tower Only Showing Shadow and Therefore it Doesn't Print

PhreakHeaven, 2023-02-27:
### Application Version

5.3.0-beta.2

### Platform

Windows 10

### Printer

Sovol SV02

### Reproduction steps

1. Add an STL that requires supports
2. Check and adjust all settings as needed
3. Find that the prime tower shadow appears, but not the actual prime tower in "Preview" mode
4. Print anyhow to see if it's just a visual bug
5. Receive a mess of soluble filament (PVA) mixed with the PLA, and consequently, a failed print

I exported the project file and opened with Cura 5.2.1 (changed nothing; simply opened the project) and the results were as expected; the prime tower existed and was visible in the slicer, and is currently printing properly.

### Actual results

No visible prime tower in the slicer and no printed prime tower...but the shadow for the prime tower is visible in Cura.

### Expected results

The slicer should generate a prime tower that would be visible in "Preview" mode and subsequently write the corresponding commands to the g-code file, causing the print to be more likely to succeed, especially when attempting to use soluble filament for the best possible overhangs.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I've loaded other models and continually experience the same issue with this version of Cura.

[image]

[SSL_Frantic Sango (7).zip](https://github.com/Ultimaker/Cura/files/10838685/SSL_Frantic.Sango.7.zip)

GregValiant, 2023-02-27:
This is a definition file issue.  This line in your "SV02.def.json" file has a typo and the syntax is incorrect.
"prime_tower_min_volume": { "value": "((reveOrValue('layer_height'))/2" },
it should be...
"prime_tower_min_volume": { "value": "resolveOrValue('layer_height')/2" },

Here is the model sliced in 5.3 beta after fixing the definition file.
[image]

I'll go ahead and remove the bug label and close this as it is a definition problem and the defintion was originally provided by Sovol.

closed 2023-02-27 by GregValiant

PhreakHeaven, 2023-02-28:
@GregValiant 

> This is a definition file issue. This line in your "SV02.def.json" file has a typo and the syntax is incorrect.
> "prime_tower_min_volume": { "value": "((reveOrValue('layer_height'))/2" },
> it should be...
> "prime_tower_min_volume": { "value": "resolveOrValue('layer_height')/2" },

Hey, thanks so much for the quick resolution and explanation; it's fixed!