Ultimaker/Cura #14711: Fuzzy Skin with User specified Z-Seam

damiandudycz, 2023-02-27:
### Application Version

5.2.2

### Platform

macOS

### Printer

Voron 2.4

### Reproduction steps

Slice model with Z-Seam set to user specified / back, and Fuzzy Skin enabled.

### Actual results

Z-Seam is placed in kinda more random places, instead of forming straight vertical line. I think this is because fuzzy skin alters the shape, and seam placement is calculated after that, but it makes this line a lot worse looking. Would be nice to be able to keep the straight line of z-seam.

### Expected results

Have z-seam placed in the straight vertical line, like it would be placed with Fuzzy skin disabled.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

MariMakes, 2023-03-21:
Hey @damiandudycz,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

When I first saw your report I thought it might be related to the Variable Line Width Engine introduced in Cura 5.0.
But this is a screenshot of 4.13.1, before it was introduced, and the zseam is also random on Fuzzy Skin, so if it's a bug it's definitely not a new bug. 
[image]

Can you help me understand why it's important to have a neat zseam if you are using fuzzy skin?
Do you have an example of an actual printjob that shows that it's important to have an aligned zseam when using fuzzy skin? 
We can use this extra information to decide on if this is more important to fix than other open issues. 
Thank you 😇

damiandudycz, 2023-05-09:
Hey @MariMakes . Sorry for delayed response. Actually it makes a difference in my situation. Im using very delicate variant of Fuzzy Skin - thickness 0.05, density 5.0. This gives parts very delicate structure, kinda like after printing with Carbon Fiber filaments. As you can see in the picture attached, it is kinda visible then
https://www.icloud.com/iclouddrive/022eVkHUvEB7e4BALxX7G5aaQ#IMG_7712

MariMakes, 2023-05-09:
Hey @damiandudycz,

Thanks for the update 👍 
We made some improvements in Cura 5.3 for the Zseam.
Could you check if you get the a better result if you change the Seam Corner Preference to None?

MariMakes, 2023-05-12:
Hey @damiandudycz,

Thanks for the update.
To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it? 

Thanks 🙏

damiandudycz, 2023-05-13:
Guess you could close it. I would rather if it allowed to create a straight line, but I guess it's more like a feature request, than bug.

closed 2023-05-15 by MariMakes