Ultimaker/Cura #14716: Slicing failed with no error

N6REJ, 2023-02-28:
### Application Version

5.3.0-beta 2

### Platform

windows 10

### Printer

Aquila S2

### Reproduction steps

Attempt to slice my model
[8-inch_GnomeTinkerer-DecorativeBase.zip](https://github.com/Ultimaker/Cura/files/10847058/8-inch_GnomeTinkerer-DecorativeBase.zip)

### Actual results

trys to slice for a long time then fails

### Expected results

Sliced properly

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

no log created 
[image]

MariMakes, 2023-03-08:
Hey @N6REJ,

Thanks for your report 👍

[image]

I took a look at your model, but this mesh seems faulty
Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/, or https://www.nanodlp.com/repair-3d-stl-file/.
Or you can use the "Make Solid" feature in AutoDesk MeshMixer. https://meshmixer.com/download.html

N6REJ, 2023-03-08:
it's strange that it didn't throw an error.

MariMakes, 2023-03-09:
Hey @N6REJ,

Perhaps I wasn't clear. I did receive your slicing error.
I believe it's related to the fact that the model isn't watertight instead of that something is wrong in Cura.
Does your model slice successfully when you make it watertight?

N6REJ, 2023-03-09:
when I did the repair yes.

MariMakes, 2023-03-10:
Awesome. That's great to hear!

To me, it doesn't seem like a bug in Ultimaker Cura, since the mesh is faulty and Cura is not built to alter meshes in this way, that's what CAD software is for.  I'll close this issue. 

Happy printing!