Ultimaker/Cura #14721: print sequence

brianable63, 2023-02-28:
### Application Version

5.2.2

### Platform

windows 10

### Printer

Modix Big 60

### Reproduction steps

Trying to turn on print sequence option,   can select print sequence in special modes, but it doesn't allow print sequence options to be set.

### Actual results

nothing.  I expect the option to become available in the list so I can change settings, but it doesn't appear in options list.

### Expected results

expect that the print sequence option shows up and I can change print sequence settings.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

test

GregValiant, 2023-03-01:
Thanks for the report.
If the printer has more than one extruder "enabled" then the setting isn't available.  Other than that it looks like it should be there.
I'm not familiar with your printer.  Is it a multi-extruder?

brianable63, 2023-03-02:
Yes, a modix 6 with two extruders.

Thanks for the response.

Brian

GregValiant, 2023-03-02:
I'm sure much thought went into making "Print Sequence" unavailable for multi-extruder printers as UM builds multis as well.  You need to disable one of the extruders and then the setting will show up.

This is "expected behavior" so I'll go ahead and remove the bug label and close this.  I hope you understand.

closed 2023-03-02 by GregValiant