Ultimaker/Cura #14723: Meshmixer to cura fail

aasmith515, 2023-03-01:
### Application Version

5.2.1

### Platform

Mac IS

### Printer

Mingda magician pro

### Reproduction steps

recently learned plane cutting on mesh mixer and using it to make armor pieces fit my Mingda since its a big printer. However soon as I try to slice it for Mingda, get Slicing error, cannot slice. I did same file in different printer (anycubic plus just rotated to fit) and it was able to slice that. 

To test it further, I did various angles back on Mingda with same file but the Magician pro would keep failing it. 
[MMPRO_back left 1.3mf.zip](https://github.com/Ultimaker/Cura/files/10855920/MMPRO_back.left.1.3mf.zip)

### Actual results

Inability for Cura to slice file at all for Mingda Magician Pro setup. 

### Expected results

Expected to see the file work so I could resume large project that has been having a lot of stop and go for long time. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

file size is cut in half from original if that helps.

aasmith515, 2023-03-01:
Also please forgive any lack of information or errors I myself cause. Very new to cad usage and printing in general, and still learning all the fine details.

GregValiant, 2023-03-02:
Thanks for the report and the file.
It took a while to slice and then failed at the end.  I checked the model and if it has any errors, they are very minor.  It seems that it should slice.
Moving and rotating the model did not help.
I'll mark this as a slicing error.
@MariMakes is on the Cura team and I think she will be interested in this one.  She uses different tools than I do and may find something
I'll mark this as a slicing error.

FYI I see you have the Support Overhang Angle set to 70°.  The maximum angle is dependent on the layer height and line width but 63° is a good rule-of-thumb for the maximum you can safely go.  Above that the individual extrusions don't bond together and prints can fail because there is no support for the overhang.

aasmith515, 2023-03-02:
Glad it isn’t just me seeing the fail! What’s weird is I tried it again yesterday for the lulz and when I raised the height of where to plane cut  it would take this new version, and sliced correctly after. 

And great suggestion on the support overhang, had no idea other settings affected other areas so much like that.