Ultimaker/Cura #14724: Incorrect cubic subdivision infill on two-material print after merge/ungroup/undo

jdulrich, 2023-03-01:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Custom IDEX printer (Tenlog TL-D3 Pro)

### Reproduction steps

Import 2 related STL files to use in a 2-material print Triangular Pivot Plate A-PETG.STL and Triangular Pivot Plate A-TPU.STL
Select Triangular Pivot Plate A-TPU.STL model and click Material 2 to assign second material to this object
Ctrl-A to select all models
Right-click and Merge Models to align both objects
Slice
Observe correct slicing behavior (on further testing, the error sometimes occurs at this point)
Right-click merged models and Ungroup Models
Ctrl-Z undo to undo the ungroup
Slice
Observe incorrect slicing behavior with different time estimate than previous

As I was writing this bug report, I noticed that when I open the project, the slicing bug occurs immediately on the first attempt. I also noticed that when I ungrouped the models that were imported from the project file, I could not merge them again properly. I don't know if that is expected behavior so I included the original STL files.

I did more experimenting and now I can't seem to get these models to slice correctly no matter what I do including restarting Cura. I can work around by changing infill pattern to Cubic so it seems the issue is with the Cubic Subdivision infill pattern specifically. It also seems like the issue may be somehow cumulative(?) and may not occur the first few times it is attempted, but may require multiple (many?) attempts to reproduce. I had sliced this model quite a few times before I noticed an issue, but the issue ruined a print.

### Actual results

Incorrect slicing behavior including multiple retracements over infill lines and infill lines where they should not be. You can see it on any layer with infill.

### Expected results

The resulting gcode should match what was output in the earlier slice

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Slicing Bug.zip](https://github.com/Ultimaker/Cura/files/10856511/Slicing.Bug.zip)

Note the extra triangular pattern in the center. I included some gcode with the error that also shows multiple retracements of the same lines which you can't see in a static image of the preview pane.
[image]

MariMakes, 2023-03-21:
Hey @jdulrich,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

Do I understand correctly that the bug is that sometimes too much infill is generated?
I'm able to reproduce your issue, and I also see that it sometimes vanishes. 
[image]

However, it seems quite innocent, especially since it's hidden inside of the model. 
Do you have an example of how it is affecting your actual printjob? 

It seems like changing your combing mode to Within Infill will have a much larger effect on the stringing in your model. 
All these darker blue travel lines are unretracted moves and will probably leave some stringing on your print. 
[image]

If you change the combing mode to Within Infill I expect you get a much more light blue lines indicated retracted moves. 
[image]

I'll bring it up to the team to see what they can do to improve it. 
But I expect that if it's not a quick fix that it will not get a high priority to implement. 
Fingers crossed 🤞

jdulrich, 2023-03-29:
I wrote this weeks ago and apparently forgot to hit send.
I don't know who to address, but thanks for the reply. It is a bit worse than extra infill - it is actually trying to extrude the same line multiple times on the same layer. The filament has nowhere to go on the 2nd and 3rd passes so could cause mechanical problems for the extruder. The multiple extrusion lines are short in my example, but if they happened to be longer, that seems like it could be bad.
I don't necessarily need you to fix it, and it is easy enough to work around by changing infill patterns. I wrote it up because it is intermittent and therefore might be a symptom of a larger systemic issue. Memory leak perhaps? I can't know
The specific impact to me was that at a time when I was working to optimize print settings, I was getting inconsistent slicing results with the same settings and watching my printer extrude back and forth over the same line 3 times per layer. It took me a while to understand why the time estimate was changing when I had changed nothing in the model or settings. By the time I figured out what was going on, I had done a lot of work to figure out how to reproduce it consistently so I figured I should write it up.
As I said, I don't care if it gets fixed. If I had to choose something for the devs to work on, I would LOOOOVE it if the interface was less glitchy on Windows 10. I use Cura exclusively because the slicing output is the best for what I do, but I sure do hate the interface. Since I have your attention, can I ask a question?
Since v5.0 it often happens when I'm trying to enter a value in a setting and I start to type "0.2", the interface validates the value immediately after I type the first '0', removes focus from the text box, and turns the textbox orange as a warning. Is there a bug on that yet or do I need to write one? I CANNOT be the only person that this drives crazy.
Thanks,Jason

MariMakes, 2023-04-04:
Hey @jdulrich,

Thanks for the reply.

For the values changing, I think I know what you mean. We've been collecting those reports under this ticket: https://github.com/Ultimaker/Cura/issues/12949. I've been trying to raise the priority of the ticket since we've been getting quite a number of reports for it.  

You can follow the progress here:  https://github.com/Ultimaker/Cura/issues/12949