Ultimaker/Cura #14725: "enable support brim" currently seems broken

fast2611, 2023-03-01:
### Application Version

5.3.0 Beta2

### Platform

Windows 10

### Printer

Anycubic Viper

### Reproduction steps

I sliced a model with "enable support brim" set to active
(also had support only every other layer and no outer walls active, but those dont affect the outcome)

### Actual results

see image
[image]
Bare underside with no adhesion help

### Expected results

see image
[image]
Adhesion help should generate

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

this is just a generic 3d test model so the post is valid and can be ignored
[chip.zip](https://github.com/Ultimaker/Cura/files/10859255/chip.zip)

GregValiant, 2023-03-01:
Thanks for the report.
I played with this a bit.  Up above in this report you say you have a Vyper, but the project file is with a "Kobra M" which I'm assuming is a Kobra Max.  I can't get a support brim with that printer active.  It's not the only oddity.  Switching to Tree supports didn't help and your supports aren't starting on the first layer.
I installed another Kobra Max as well as a Vyper and the problem persists with all three of the Anycubic printers.
I switched to my Ender and everything appears to be fine EXCEPT conical supports doesn't work and no support is generated when Conical is turned on.
Someone from the Cura team will take a look.  This might be more involved than it looks on the surface.

fast2611, 2023-03-01:
I just realized I didnt mention I had conical supports on as well, sorry.
As for the printer: I used a viper preset since it had the closest specs out of the available presets to the kobra max at the time I set it up, which is the actual printer I'm using, I figured i'd list what I selected in cura since the physical machine it ends up on doesnt really matter.
I scrolled through issues about an hour ago and saw a few mentioning  Conical supports being various degrees of broken, but disabling the option doesn't seem to fix the lack of support brim.
I have additionally noticed that an support interface with a extremely big resolution gets sometimes generated against the buildplate, as if it was printing onto an existing model 
Thanks for looking into it, besides the missing parts I'm happy to report that most supports generally look alot cleaner this version, so good work so far <3

fast2611, 2023-03-01:
(also accidentally closed this , my bad)

ksuquix, 2023-10-03:
Just an observation from my difficulties with this issue:  Somehow the support brim seems to be linked to regular brims now.  If you turn on regular brims, they will work (of course, this is not desirable, but may shed light on the situation).

GregValiant, 2023-10-04:
You can set the "Brim Distance" to a low number like "2.0" and the brim will be separated from the model (more like a skirt) and the Support Brim will form correctly.

fast2611, 2023-10-04:
you can do the same with an actual skirt and still get the effect, while using even less fillament most of the time, due to doing only a half loop most of the time like that (no idea why it cant be gotten lower than that)
The biggest downside is that both workarrounds take a slight amount of build volume away from what would otherwise fit on the plate, which can be compensated by making the printer bed slightly larger than the actual size in the machine settings (just dont exceed your safety clearance, please)

fast2611, 2023-10-04:
dear god I need to stop fat fingering the close issue button, I'm so sorry xD