Ultimaker/Cura #14726: Slicing failed

Greenie1000, 2023-03-01:
### Application Version

5.3.0-beta.2

### Platform

Windows 10

### Printer

Ender 5

### Reproduction steps

uploaded and stl and sliced using my personalized settings for petg. 
Cura started the assessment but then failed to complete. 
Error message below
[image]

### Actual results

Fails to complete but allows me to try to slice again. 

### Expected results

The item should have been sliced successfully. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CE5_Plopper_versie_1.zip](https://github.com/Ultimaker/Cura/files/10859944/CE5_Plopper_versie_1.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/10859979/cura.log)

GregValiant, 2023-03-01:
Thanks for the report.
I didn't have any trouble slicing this (with 5.3b) so I'm unable to duplicate the problem.
Have you tried moving the model on the Cura build plate?

Greenie1000, 2023-03-02:
Hi Greg,
You beat me to it – one I rotated the model slightly and resliced it worked straight away. First glitch I had experienced.
appreciate the follow up
Cheers
Steve

GregValiant, 2023-03-02:
Thanks for getting back.
I'll mark this as a duplicate then.  This "won't slice when in the middle" bug was worked on for 5.3b but I guess the fix isn't there yet as this is the second report I've seen here.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker