Ultimaker/Cura #14728: No 3mf thumbnail shown

GMagician, 2023-03-01:
### Application Version

5.2.2

### Platform

Windows 11

### Printer

Custom

### Reproduction steps

Whenever any 3mf containing a file with double #n (e.g. 'myprinter_extruder_0 #2_user #2.inst.cfg' windows doesn't display thumbnail. To see it just open 3mf with 7zip and rename a 'Cura/' file.

Now I've not enough knowledge to know if this is a "malformed" 3mf or a windows thumbnail viewer issue, erase this if latter

### Actual results

3mf cura icons is used

### Expected results

thumbnail image as icon

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No needs for a specific 3mf or printer configuration

MariMakes, 2023-03-07:
Hey @GMagician,

Thanks for your report 👍
I was able to reproduce it by following your exact steps. 
But not in another way.

Can you help me understand why this is a bug?
How is this limiting you or others in working with Cura?

GMagician, 2023-03-07:
I think it's a bug because thumbnail should be shown.
Question is:
Does such filename turn this 3mf file into a malformed 3mf (then windows doesn't display image) or it's just the windows viewer bugged? I don't have enough knowledge to discriminate, but someone here for sure

This bug doesn't limit anything, it's just lost the possibility to see at a glance what is contained into a 3mf.

GMagician, 2023-03-07:
Oh I forgot. Cura may create such filenames (with 2 #) when a 3mf is loaded and then saved. Usually I see it if 3mf was saved by old printer and loaded with new one. Not detected properly when this occurs. Maybe when 'groupid' changes (in global.cfg)