Ultimaker/Cura #14732: stringing

Cosplay101, 2023-03-01:
### Application Version

5.2

### Platform

Window 11

### Printer

ender 3v2 neo 

### Reproduction steps

It may have been a corrupt model I downloaded because the travel path is all over the place on all the previous models I used before using the recent model I got now no matter what I do the slicer will always show stringing I tried reinstalling and uninstalling  is there a way to completely reset the software for my laptop another instance is that while trying to fix the issue some to of the setting disappear suddenly and wouldn't come back the first picture is the one that might be corrupt model please help
[image]
[image]
[image]
[image]
 unless exit and reopen the program 

### Actual results

the stringing never stops and the solutions for it won't work program maybe corrupted 

### Expected results

Cura slicer may be broken for my computer need full reset 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

can't add files on https://www.thingiverse.com/thing:2788589/comments

GregValiant, 2023-03-01:
Thanks for the report.
That's pretty wild.  I haven't seen that before.
With the model loaded and Cura set up to slice use the "File | Save Project" command to create a 3mf project file.  Zip the 3mf file and post the zip folder here.  Someone will take a look.

Cosplay101, 2023-03-02:
I've uploaded two of the same project one that was done when it wasn't stringing and the other that is was done recently to attempt to solve the problem but fail 
the previous 
[CE3E3V2_Pen Holder With Loops.zip](https://github.com/Ultimaker/Cura/files/10870637/CE3E3V2_Pen.Holder.With.Loops.zip)
then recent 
[CE3E3V2_Pen Holder With Loops stringing.zip](https://github.com/Ultimaker/Cura/files/10870644/CE3E3V2_Pen.Holder.With.Loops.stringing.zip)
 
I've included the gcodes if that will help 
[Cura Pen Holder With Loops 230.zip](https://github.com/Ultimaker/Cura/files/10870648/Cura.Pen.Holder.With.Loops.230.zip)

[CE3E3V2_Pen Holder With Loops stringing (2).zip](https://github.com/Ultimaker/Cura/files/10870685/CE3E3V2_Pen.Holder.With.Loops.stringing.2.zip)

please if anyone can help I really am stumped It've try going back to the default setting when the stringing wasn't there but it was still there also is there anyone I could get help Irl I almost feel like I might need someone to look at it to find the said problem

GregValiant, 2023-03-02:
In your second screenshot in the original post you can see the "Color Scheme" dialog box.  Turn off "Travels".
Here is your "good" file but with travels turned on.
Is this what you were seeing?  If there is an extrusion during a travel move Cura shows that as a heavy blue line rather than the thin cyan line of a regular travel move.
[image]

Here is your "bad" file with Travels turned off.
[image]

Cosplay101, 2023-03-02:
Okay so I may have a dumb question I heard when you turn combing on those travel line disappear sorry I’m just a little confused about the travel lines on stringing or it’s just depends on printer

[image]<https://user-images.githubusercontent.com/64202104/222447232-99ed8110-7cc5-4f9f-a09a-150645eccd12.png>

[image]<https://user-images.githubusercontent.com/64202104/222447609-9739da99-add1-47ab-a5c0-bbf8a7883e23.png>

GregValiant, 2023-03-03:
The travel lines never disappear.  With some combing options the travel lines are hard to see because they follow walls (or whatever) but they are there.
When you have a model like you are showing that separates into "islands" as it gets taller then there must be travel moves between the islands.  If the printer is nicely dialed in there will be none-to-minimal "stringing" on the print.  Combing doesn't really work for "islands" as there are no features for the nozzle to follow when it is traversing the air between islands.  Think of that model as 6 separate models once the features split up.

Cosplay101, 2023-03-05:
Thanks sorry I had took a break from 3d printing for awhile and must of turn the travel on layer view I think it’s usually off when I first installed it thanks for the information