Ultimaker/Cura #14733: Cura slicing failed

mmcp42, 2023-03-01:
### Application Version

5.1.0

### Platform

windows 11

### Printer

Creality 3

### Reproduction steps

made a model in Fusion 360
attempted to slice it
was ok with no supports
added tree supports
"something went wrong"
tried again with normal supports
works ok

### Actual results

Cura just aborted the slicing

### Expected results

was expecting it to slice
[grip.zip](https://github.com/Ultimaker/Cura/files/10864740/grip.zip)
if you want more ino
mmcp42@yahoo.com

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10864780/cura.log)

mmcp42, 2023-03-01:
[grip.zip](https://github.com/Ultimaker/Cura/files/10864788/grip.zip)
updated zip file

MariMakes, 2023-03-17:
Hey @mmcp42,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

The new Cura 5.3 contains the fix for this for a number of slicing crashes. 
It seems to fix your error aswell!
[image]

You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
Would you still like to keep this issue open or do you agree that we can close it?

mmcp42, 2023-03-19:
Hello there

Yup seems only fair

It’s working now

So bug sorted

Please do close it

cheers

 

Warmest regards

 

Mike McPherson

07711 238858

***@***.***

 

 <https://user-images.githubusercontent.com/40423138/225973201-5aaac4c4-5aee-458a-b5a3-5fca23ef753d.png>

MariMakes, 2023-03-20:
Good to hear! Happy printing! 
Feel free to let us know if you run into anything else.

closed 2023-03-20 by MariMakes