Ultimaker/Cura #14736: One Nozzle, Two Extruders/Tools.  Missing second heat settings in g-code.

jurban1997, 2023-03-01:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

SeeMeCNC Boss300Delta

### Reproduction steps

When printing the following warning message occurs and the heater shuts down:  "Warning: Tool 1 was not driven because its heater temperatures were not high enough or it has a heater fault".  
The printer is a chimera at the head.  There is only one nozzle and two drivers (T0 & T1).  During the filament switch it retracts and the next one is driven while the heater adjusts to that extruder's heat setting.  
When it switches to T1 there is no heat setting (M104, 109 or 190).  The thermistor senses the drop in heat and stops driving the extruder tool.  

### Actual results

In the G-Code, only the first layer's tool is added to the initial section:

```
;Generated with Cura_SteamEngine 5.2.1
T0
M190 S65
M104 S220
M109 S220
```

### Expected results

The G-Code needs to look like this:
```
;Generated with Cura_SteamEngine 5.2.1
T0                                                               ; Tool 0 heat settings
M190 S65
M104 S220
M109 S220
T1                                                               ; Tool 1 heat settings
M190 S65
M104 S220
M109 S220
```

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Product:  https://www.seemecnc.com/products/bossdelta-300-3d-printer

GregValiant, 2023-03-02:
Thanks for the report.
Load the "Printer Settings" plugin from the Cura Marketplace.  After restarting Cura, set your "Setting Visibility" to "All".  Printer Settings will be a new section below "Experimental".
Scroll down through the Printer Settings and enable "Extruders Share Heater" and "Extruders Share Nozzle".  That should fix the problem.
On a dual hot end multi-extruder printer - Cura does not heat a tool until its going to be switched to.  When the tool change occurs from T0 to T1 Cura sets T0 to the material Standby Temperature.  Since you only have 1 heater it cools off and T1 doesn't work.
Let me know if that fix works for you as this sounds like "expected behavior" given your 2-in-1-out hot end.
Right now your gcode probably looks like this at a tool change:

G1 F3000 X39.419 Y174.052 E619.79753
G1 F1200 E603.79753
G92 E0
**T1**
G92 E0
M105
M109 S200
**M104 T0 S175** ------Here is the problem.  This line will go away when you enable "Extruders Share Heater".

jurban1997, 2023-03-02:
Installed the Printer Settings plugin.  Enabled  "Extruders Share Heater" and "Extruders Share Nozzle".  Ran the slice and reviewed the G-Code. 

At the top there is an M190 S<temp> line for the buildplate.  
There were no M104 lines until the end where it was:
`M104 S0`  to turn it off.

Even more strange, there was no S<temp> settings anywhere with the printing temperature I specified.  How would the printer know what temp to print?

Another oddity was that the Print Temperature setting disappears from Extruder 1 on the sidebar.  It can be edited on Extruder 2, but that was a bit confusing until I discovered it.  And, logically, I might want to change the temp for each extruder anyway. So it's a bit of an assumption that both materials would always use the same temp.

I've uninstalled the plugin and reverted to the prior profile and I'm back to where I started.  

Shouldn't this (eventually) be a setting on the Dual Extrusion group?