Ultimaker/Cura #14738: Slicing error until transformed

GiulianoCTRL, 2023-03-02:
### Application Version

5.3.0-beta.2

### Platform

MacOS

### Printer

Creality Ender-3 S1 Pro

### Reproduction steps

1. Load project
2. Try slicing model -> Fails
3. Change scale of models by 1%
4. Slice once more -> success

### Actual results

Slicing fails around 30% through the process.

### Expected results

Slicing should work without needing to transform the model.
NOTE: Slicing works without issues for other models, there is just a few, usually round models that seem to cause this issue.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Probably related to https://github.com/Ultimaker/Cura/issues/14734 , but it seems there isn't enough information attached.

[cura.log](https://github.com/Ultimaker/Cura/files/10871568/cura.log)
[Slicing_error_project_file.zip](https://github.com/Ultimaker/Cura/files/10871570/Slicing_error_project_file.zip)
[image]
[image]

GregValiant, 2023-03-03:
Thanks for the report and the files.
This project sliced for me on the first try so I'm unable to duplicate the problem.
Have you tried moving the models on the build plate and/or rotating them?
I'll mark this as a duplicate slicing error.  Someone from the Cura team will take a look.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker