Ultimaker/Cura #14739: housing with modellized threads

AP-Reinhard, 2023-03-02:
### Application Version

5.2.2.

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

Loaded the stl-file
fixing printing options, using support structure
start slicing
[image]
[image]
[image]
[image]

### Actual results

Slicing does create failure

### Expected results

correct slicing

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[UMS5_Gehäuse 01-PR20025-05_Druckversion.zip](https://github.com/Ultimaker/Cura/files/10871923/UMS5_Gehause.01-PR20025-05_Druckversion.zip)

GregValiant, 2023-03-02:
Thanks for the report.
Your model sliced for me on the first attempt.  Since it was a model file and not a project file my Ender was the active printer.  It's not the same size as your S5 and so the model was not centered.  There is a bug in Cura that hates when the models are in the center of the build plate.  Try moving and/or rotating the model.  You may find a spot where it will slice.
I'll mark this as a duplicate report.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker