Ultimaker/Cura #14742: STL no longer slices using custom profile under 5.3.0-beta2

tisimst, 2023-03-02:
### Application Version

5.3.0-beta2

### Platform

Windows 10

### Printer

Anycubic Chiron

### Reproduction steps

Load part, scale 5x, translate z-450, slice

### Actual results

Fail to slice

### Expected results

Slice successfully (profile slices fine with recent 5.3.0-alpha+xmas).

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10875289/cura.log)
[StaffExportscale-no-slice-5.3.0-beta2.zip](https://github.com/Ultimaker/Cura/files/10875293/StaffExportscale-no-slice-5.3.0-beta2.zip)

GregValiant, 2023-03-03:
Thanks for the report and the files.
I rotated your model 45¯ and it sliced.
I'll mark this as a duplicate and a slicing error.  Someone from the Cura team will take a look.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker