Ultimaker/Cura #14745: Slicing

pinot911, 2023-03-03:
### Application Version

5.3.0

### Platform

MacOS Catalina

### Printer

Creality Ender -3 v2

### Reproduction steps

Slicing failure

### Actual results

It just fails and says report bug

### Expected results

Slices!

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Cannot attach CMF files
[hose to hose.3mf.zip](https://github.com/Ultimaker/Cura/files/10878340/hose.to.hose.3mf.zip)

GregValiant, 2023-03-03:
Thanks for the report and the file.
Your model sliced for me in 5.3beta with no changes (I'm on Windows).
As a workaround, try rotating the model and maybe moving it in Cura.  That will often fool this bug.
I'll mark this as a duplicate and a slicing error and the team will take a look.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker