Ultimaker/Cura #14747: Slicing failed with an unexpected error

rpersak, 2023-03-03:
### Application Version

5.3.0 Beta 2

### Platform

Windows 11

### Printer

Ender 3 pro

### Reproduction steps

1. Loaded model
2. clicked "slice"

### Actual results

I get an error: 
[image]

### Expected results

Sliced model

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CE3PRO_gravity_feeder_tall.zip](https://github.com/Ultimaker/Cura/files/10879923/CE3PRO_gravity_feeder_tall.zip)
[cura - Copy.zip](https://github.com/Ultimaker/Cura/files/10879925/cura.-.Copy.zip)

GregValiant, 2023-03-03:
Thanks for the report and the files.
I agree this model should slice as is.  I rotated it 90° and moved it but it still wouldn't slice.  Rotated to 45° it sliced right away so that might be your workaround.

I'll mark this as a slicing error and as a duplicate.  This bug was worked on for 5.3 but doesn't seem to be fixed yet.
[image]

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker