Ultimaker/Cura #14749: Cura sends "M105" to serial 2D printer

ikeji, 2023-03-03:
### Application Version

5.3.0-Beta2

### Platform

Ubuntu 20.04

### Printer

Ender3 + Otoprint

### Reproduction steps

I have *2D* Receipt printer and attached to /dev/ttyS0.
https://www.adafruit.com/product/597

### Actual results

When I start cura, the printer keep prints "M105" repeatedly.

### Expected results

The cura doesn't touch the port that is not specified.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10881289/cura.log)

https://twitter.com/ikeji/status/1631639132899540993

GregValiant, 2023-03-03:
Thanks for the report.
The USB Printing plugin can be disabled in the Cura MarketPlace.  That's your best option as you are using Octoprint.
Since USB Printing is a legacy plugin it isn't maintained by anyone but it continues to be included with Cura because some folks like to print via USB.

I'll remove the bug label from this.  It's expected behavior from "USB Printing".  Let me know if you feel differently about that.

ikeji, 2023-03-03:
Thanks for reply.

I confirmed that when I disabled the plugin, there are no unexpected print.

Thanks,

GregValiant, 2023-03-04:
The plugin sends M105 and waits for a response from the printer.  If it understands the response then it knows the baud rate is correct.  Since there was no response, or a garbled response from the receipt printer USB Printing just kept trying and the printer kept printing.
I'll go ahead and close this then.

closed 2023-03-04 by GregValiant