Ultimaker/Cura #14750: Fuzzy Skin problem

Raymo99, 2023-03-03:
### Application Version

5.3.0 beta 2

### Platform

Windows 10,  64 bit

### Printer

Ender 5

### Reproduction steps

Set Fuzzy skin to default on a cube

### Actual results

Only three sides are fuzzy, the 4th side is only slightly fuzzy... as if only some layers get fuzzy

### Expected results

all 4 sides should be Fuzzy

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2023-03-04:
Thanks for the report.
I can duplicate this condition.  Someone from the Cura team will take a look.
[image]

MariMakes, 2023-03-07:
Hey @Raymo99,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍
That definitely looks like a bug that slipped through! 🐛
It looks like the last edge is not "fuzzyfied" 😞 

I don't think a fix will make it stable, but we are working on a solution. 💪 
For internal reference CURA-10368

Raymo99, 2023-03-08:
Hi Mari,
Glad it was easy to reproduce.

Chippik, 2025-01-14:
hello? i faced the same problem. one side of my model is not fuzzy. could you reveal the solution?

Amador1999, 2025-03-10:
when fix this problem???

GregValiant, 2025-03-10:
I just checked a Benchy and a cube in 5.9.0 and I don't see the problem.
What version are you talking about?

Amador1999, 2025-03-11:
5.9.1

Cu stima  Ing D.Tanase
Tel.:0722686411

În lun., 10 mar. 2025 la 22:30 GregValiant ***@***.***> a
scris:

GregValiant, 2025-03-11:
Files cannot be posted on Github via email.  They must be posted directly to the site (Github rules).
I just wanted to confirm that this problem is still present in recent versions.  I wasn't able duplicate the problem now IN 5.9.0 but I could in 5.3.1.
If someone would post a project file I will take another look.