Ultimaker/Cura #14751: Missing Support

LBowers23, 2023-03-03:
### Application Version

5.3.0-alpha+xmas

### Platform

Win 10

### Printer

Ender 3 Pro

### Reproduction steps

Loaded model (stl in zip file)
Rotated 180 deg so gears on build plate
set slicer to include supports
sliced
NO SUPPORT ON INTERIOR OVERHANG IN SHAFT IS GENERATED (towards bottom)
[test gear with key.zip](https://github.com/Ultimaker/Cura/files/10883920/test.gear.with.key.zip)

### Actual results

The inner overhang is printed into unsupported air
[image]

### Expected results

Should be supports under overhang

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Tested on Cura 5.1.0 and supports are there

GregValiant, 2023-03-04:
Thanks for the  report and the file.
Because that step is 0.55 wide you will have to make some serious adjustments Cura to support to it.  
The screenshot below is with a 0.2 nozzle.  I could not get the support to generate with a .4 or .3 nozzle.  Because the ledge is so narrow you cannot use a Support Interface.
I also had to set the support XY distance to 0.05.  That will essentially bond the support to the model.
[image]

Going the other way on nozzle size - a 0.6 nozzle at 0.6 line width would not require support for that feature.

I'm inclined to remove the bug label from this.  Support can be generated there, it just isn't easy.  Let me know what you think.

LBowers23, 2023-03-04:
Your diagnosis makes sense.  And if you'd like the 'Bug' designation can be removed.

Perhaps between the 5.1 and XMAS versions I have a setting that is affecting this support from not being generated.  Thank you for looking further.

GregValiant, 2023-03-04:
I didn't have any success getting support in 4.13.1 either.  I think I'll leave the bug label on and let someone from the Cura team look at this.  They are working on supports for the 5.3 stable release and this may be of some interest to them.

LBowers23, 2023-03-04:
Appreciate your looking.  No biggie for me.  With a some rotary tool fixing I can make it work.  This was part of a Thingiverse file I was looking at for a project.

Larry