Ultimaker/Cura #14753: Cura causing Samsung G9 Odyssey to flicker

usnavyitc, 2023-03-03:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Ender 3 Pro

### Reproduction steps

1. Open Cura.
2. Cura opens but then a few seconds later, the monitor goes black as if changing inputs then comes back. Does this constantly. Only way to stop it, is to close Cura.
3. This is only happening when opening Cura. If I connect a second monitor and move the Cura instance to that monitor there are no issues however if I keep Cura open on the 49" G9 Odyssey, it becomes unusable as the screen will constantly go black.

### Actual results

At this time, Cura is unusable on this monitor. 

### Expected results

Cura should be able to work on any monitor. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have confirmed I am running latest video drivers (v531.18) on my NVIDA 2080 TI. Samsun G9 Odyssey is set to 5120x1440 at 240Hz. I attempted to use previous versions of Cura with no issue. I tested using PrusaSlicer and no issue. Cura is the only application that doesn't seem to want to work. No issues with any other application.

MariMakes, 2023-03-21:
Hey @usnavyitc,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

We have a range of workarounds for people experiencing trouble with Cura on Monitors.
Could you check if any of these are relevant to you? 

1.. Do you have two monitors placed above each other? Can you try this workaround from the community? https://community.ultimaker.com/topic/42633-cura-522-stopped-working/#comment-318105
2. Do you have GPU Tweak OSD overlay enabled? Does Cura start if you disable that software?
3. Do you have a Ryzen Graphics card? Can you try this workaround? https://support.makerbot.com/s/article/1667412437627
4. Do you have the Nahimic audio enhancement service enabled? Can you try this workaround? https://github.com/Ultimaker/Cura/issues/12102#issuecomment-1157907665

If this doesn't resolve your issue, let me know and I'll take another look 👍

MariMakes, 2023-03-31:
Quick update on our side

We might have a fix for some of the issues reported here. 🎉
It's a bit hard for us to test, but we have a nightly build available.
Can you please try it out and let us know if it resolves your issue?

[You can find the download links and the discussions here.](https://github.com/Ultimaker/Cura/discussions/15098)