Ultimaker/Cura #14755: Slicer error

MartinNovan, 2023-03-03:
### Application Version

5.3.0 - beta 2

### Platform

Windows 10 pro

### Printer

Ender 3

### Reproduction steps

I load my print profile with layer heigth 0.08mm and i does not slice, so I changed it to default profile that has layer heigth 0.12mm and it worked. So i bellieve that there is an error in slicing when the layer height is 0.08mm because i tried to change the layer heigth in the default profile too and got the same result.

### Actual results

It´s working with 0.12mm layer heigth but not with 0.08mm height.

### Expected results

In the preview version the layer heigth normally work with 0.08mm.
[CE3E3V2_headphone_holder_56mm.zip](https://github.com/Ultimaker/Cura/files/10885581/CE3E3V2_headphone_holder_56mm.zip)
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I cannot find the .log file in folder and throught cura the button didnt work so i will attach a screenshot of my cura folder.
[image]

GregValiant, 2023-03-04:
Thanks for the report and the Project file.
It would not slice for me either.  My log file showed a memory address error _"Backend exited abnormally with return code 3221225477!"_
I changed the "Initial Layer Height" to 0.14 and it sliced.
I think this is a duplicate of other reports where slicing failed and a minor adjustment to the initial layer height fixes it.  I noticed that moving the model on the Cura build plate also allowed it to slice.
I'll mark this as a duplicate and a slicing error and the Cura Team will take a look.

The "Cura Log File" is a text document.  You have Windows "Hide file name extensions of known types" turned on but it appears the "cura...textovy dokument" is the log file.

You probably noticed, but with your settings that's a 37 hour print.

Vandresc, 2023-03-06:
Fixed for the 5.3 stable, just gave it a try with your model, and it successfully slices. Thank you for reporting it!

closed 2023-03-06 by Vandresc