Ultimaker/Cura #14760: Cura 5.2.1 incorrecly generated support (not respecting support placement build plate only)

ogsadmin, 2023-03-04:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ultimaker S3

### Reproduction steps

1. Place a model (STL file and 3mf project attached, zipped: [Documents.zip](https://github.com/Ultimaker/Cura/files/10889037/Documents.zip))
2. Set Extruder 1 as support extruder (PVA)
3. Set Extruder 2 as print extruder (PLA or ABS, irrelevant)
4. Slice
5. Check the model in the preview (material color scheme) --> it shows support extruder 1 putting material at places where no overhangs are/no supports are needed (inside holes, at edges) - see attached screenshot: 
[image]

### Actual results

The sliced model shows unneccessary support extrusions (as shown in the screenshot):
[image]

The setting "Support Placement" = "Touching Buildplate" seems to be not respected, as support is generated at places, which do not have contact with the build plate.

### Expected results

There should only be support where needed and only if the support touches the build plate.
It does seeminly not make sense to have support material extruden on the top of the model.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Log file and cura.cfg (zipped): [5.2.zip](https://github.com/Ultimaker/Cura/files/10889052/5.2.zip)

GregValiant, 2023-03-04:
Thanks for the report.
The 3mf file opened as a regular model file rather than a project file so I'm not sure I'm on the right track here.
Those white spots appear to be the starts and stops of the outer wall extrusions (the z seam).  You can turn them off in the Color Scheme dialog.  Supports are cyan colored.
This is your file sliced with "Starts" turned on.
[image]

This is with Starts turned off.
[image]

I'll leave the bug label on this for now but it seems to be a Preview setting rather than a bug.  Let me know what you think.

ogsadmin, 2023-03-05:
Hi Greg,

sorry - you are right. I had my color scheme settings set to "Material color" and this showed the starts in white (i.e. similar to the PVA color, so this was barely noticable). Here is a screenshot:
[image]
Maybe a bit confusing is that even though the "Starts" entry is listed (and checked) it does not show a color next to it - it seems to default to white here...

If I uncheck the "Starts" entry in the "Material Color" scheme or switch to the "Line type" color scheme, everything is as expected:
[image]

Sorry for the wrong bug report - and thanks for analyzing!

Please go ahead and close it.

GregValiant, 2023-03-05:
Not a problem, it's an easy thing to miss when you are concerned about 300 other settings.

closed 2023-03-05 by GregValiant