Ultimaker/Cura #14764: Cura 5.3.0-beta 2 slicing bug

Andrethetech, 2023-03-04:
### Application Version

5.3.0-beta.2

### Platform

Windows 11

### Printer

Anet ET4

### Reproduction steps

Sliced a Benchy

### Actual results

There are missing layers on the bottom of the cabin see picture.

### Expected results

Proper slicing

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Here are the files 

[Cura 5.3.0 beta 2 bug.zip](https://github.com/Ultimaker/Cura/files/10889286/Cura.5.3.0.beta.2.bug.zip)

GregValiant, 2023-03-05:
Thanks for the report.
@MariMakes this is easily duplicated.
Below is a screenshot from 4.13.1 using OP's project file.
[image]

I opened the OP's project in 5.3beta and the problem showed up (it looked like his screenshot).
In the screenshots below I have changed the active printer to my Ender and my own setting profile.
My Double-Ender Benchy is at the top, the OP's Benchy is in the middle, and my SportFish Benchy is at the bottom.  I turned on supports because it makes this really easy to see.
Here are the boats.  There were errors in the OP's model so I repaired it in MS 3D Builder.  There are no errors in these models.
[image]

This slice is in 5.3beta.  There is 2.2mm of structure missing from the two boats while the one I combined appears to be OK.
[image]

Here is a shot from 5.2.1 and everything looks as it should.
[image]

@fieldOfView am I missing something here?

MariMakes, 2023-03-07:
Hey All! 

Thanks for flagging this 💪 
That's definitely a bug that slipped through 🐛 

It seems to have been related to the Hole Horizontal Expansion.
@casperlamboo was able to fix the issue before our stable release here https://github.com/Ultimaker/CuraEngine/pull/1857
Thank you for reporting ❤️

closed 2023-03-07 by MariMakes