Ultimaker/Cura #14773: Crash on custom profile rename

neut, 2023-03-05:
### Application Version

5.2.2

### Platform

Windows 11

### Printer

Ender 3v2

### Reproduction steps

1. Profile --> Manage profiles
2. Select custom profile ("Not supported #2")
3. Burger menu --> Rename
4. Enter any arbitrary string
5. Click OK
6. Crash

### Actual results

Cura crashes. See screenshot/log
# Screenshot
[image]

### Expected results

Expect rename of custom profile

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

# Traceback
Traceback (most recent call last):
  File "cura\Machines\Models\QualitySettingsModel.py", line 70, in setSelectedQualityItem
    self._update()
  File "cura\Machines\Models\QualitySettingsModel.py", line 114, in _update
    global_containers = container_registry.findContainers(id = metadata_for_global["id"])
KeyError: 'id'

# Log
Thread 0x00004a68 (most recent call first):
  File "C:\Users\neut1\AppData\Roaming\cura\5.2\plugins\OctoPrintPlugin\OctoPrintPlugin\python-zeroconf\src\zeroconf\_services\browser.py", line 541 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000423c (most recent call first):
  File "asyncio\windows_events.py", line 788 in _poll
  File "asyncio\windows_events.py", line 439 in select
  File "asyncio\base_events.py", line 1860 in _run_once
  File "asyncio\base_events.py", line 600 in run_forever
  File "asyncio\windows_events.py", line 321 in run_forever
  File "C:\Users\neut1\AppData\Roaming\cura\5.2\plugins\OctoPrintPlugin\OctoPrintPlugin\python-zeroconf\src\zeroconf\_core.py", line 494 in _run_loop
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001bc4 (most recent call first):
  File "C:\Program Files\UltiMaker Cura 5.2.2\share\cura\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000052b0 (most recent call first):
  File "threading.py", line 324 in wait
  File "zeroconf\__init__.py", line 2534 in wait
  File "zeroconf\__init__.py", line 1715 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000059e0 (most recent call first):
  File "threading.py", line 324 in wait
  File "threading.py", line 600 in wait
  File "C:\Program Files\UltiMaker Cura 5.2.2\share\cura\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00005a1c (most recent call first):
  File "zeroconf\__init__.py", line 1369 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00002c2c (most recent call first):
  File "C:\Program Files\UltiMaker Cura 5.2.2\share\cura\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001f08 (most recent call first):
  File "UM\Backend\Backend.py", line 169 in _storeStderrToLogThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00005bd8 (most recent call first):
  File "UM\Backend\Backend.py", line 157 in _storeOutputToLogThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001f88 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00002ff0 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00003dfc (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00002d78 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00002ca8 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001780 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000041cc (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00004928 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00005b20 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00002464 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000257c (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000050ac (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Current thread 0x00004f08 (most recent call first):
  File "cura\CrashHandler.py", line 397 in _logInfoWidget
  File "cura\CrashHandler.py", line 182 in _createDialog
  File "cura\CrashHandler.py", line 86 in __init__
  File "cura_app.py", line 165 in exceptHook
  File "UM\Qt\ListModel.py", line 115 in setItems
  File "cura\Machines\Models\QualityManagementModel.py", line 399 in _update
  File "cura\Machines\Models\QualityManagementModel.py", line 279 in _qualityChangesListChanged
  File "UM\Signal.py", line 357 in __performEmit
  File "UM\Signal.py", line 323 in __handleEmitIndirect
  File "UM\Signal.py", line 216 in emit
  File "UM\Settings\ContainerRegistry.py", line 841 in _onContainerMetaDataChanged
  File "UM\Settings\InstanceContainer.py", line 218 in setName
  File "cura\Machines\Models\QualityManagementModel.py", line 134 in renameQualityChangesGroup
  File "UM\Qt\QtApplication.py", line 416 in exec
  File "cura\CuraApplication.py", line 897 in run
  File "cura_app.py", line 239 in <module>

MariMakes, 2023-03-07:
Hey @neut,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Sorry to hear you are having a crash. 
The profile naming has changed in Cura 5.3.
Are you still experiencing the crash in there? 
[image]

If that's not it, can you share how you want to rename the profile?
We might also need your configuration folder with profiles to reproduce it. 

You can share your configuration folder by:
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version
You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

neut, 2023-03-10:
Thanks for the response :) I tried Cura 5.3. Unfortunately, the problem persists.

[image]