Ultimaker/Cura #14774: print in ultimaker cura vs irl

mieleken12, 2023-03-05:
### Application Version

5.2.0

### Platform

windows 10 i think

### Printer

ender 5 (pro)

### Reproduction steps

1.so as normaly i opend up ultimaker cura and i wanted to print ditto (not only ditto it happend before) and i uploaded it on my sd card
2.i went to my ender 5 (pro) and loaded it up. i put in the sd card and started printing
3. i went to wash myself and came back and it was a mountain of spaghetti
4. and it was not the first tim
5. pls help me

### Actual results

um i write it up here sorry i did not read everything

### Expected results

i dont want spaghetti anymore and i dont know the settings that well so maybe a little bit more clearly

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

i dont know how to do it

GregValiant, 2023-03-05:
Sorry to hear of your problem.  This isn't a support site but rather the UltiMaker bug reporting hub.  Have you tried asking for help in places like the [Reddit Creality Ender 5](https://www.reddit.com/r/ender5/) group?
I'm going to remove the bug label and close this as it appears to be a support issue for a Creality printer.  I hope you understand.

mieleken12, 2023-03-05:
thank you so much i get it i went a little to far in the story. i hope you
understand. stil thank you for helping me.😀

GregValiant, 2026-09-10:
I'll close this.

closed 2026-09-10 by GregValiant