Ultimaker/Cura #14775: Ultimaker Cura 5.3.0 Beta 2 Adhesion Bug

tahayusab, 2023-03-05:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Anycubic i3 Mega S/P/X

### Reproduction steps

1. I added text cube
2. added brim adhesion 
3. sliced
4. no brim was shown in the preview

### Actual results

no brim was shown in the preview

### Expected results

seeing the brims or any types of adhesion in the preview section

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2023-03-05:
Thanks for the report.
I know I can't duplicate this as the print I have running now has a brim.  That's why we ask for Project Files.
Check your preview "Color Scheme" and see if you have "Helpers" disabled.

tahayusab, 2023-03-06:
yeah you were right in the old version because i didn't changed it i didn't think about it. thanx for your help

closed 2023-03-06 by tahayusab