Ultimaker/Cura #14776: The part changed the scale to 10000% automatically without my interference.

Mkanton, 2023-03-05:
### Application Version

5.2.2

### Platform

Windows 10 22H2

### Printer

Ultimaker S3

### Reproduction steps

1. I loaded the file normally as I do with other files.
2. The piece is 10,000% larger than the original.
3. Even when I tried to change the scale to 100% the program didn't change the scale.

### Actual results

I was unable to print the part. I had to carry another piece.

### Expected results

I expected the piece to print normally like every time.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10891948/cura.log)
[PROJETO - BAIXO  com Settings.zip](https://github.com/Ultimaker/Cura/files/10891949/PROJETO.-.BAIXO.com.Settings.zip)

GregValiant, 2023-03-05:
Thanks for the report.
You seem to have a bad STL file there.  I tried to open it in Mesh Mixer, IdeaMaker, PrusaSlicer, and MS 3D Builder.  They all reported that it is an invalid file.
I'm going to remove the bug label from this but I'll leave it open for a bit so you can respond here.  I hope you understand.

Mkanton, 2023-03-06:
Hello Greg! I'm afraid you're right. After your answer, I generated a new STL, then I was able to print normally.
You mentioned that you opened the file in the Prusa Slicer. I would love to try this slicer on my Ultimaker 3S printer. Do you know how I configure the printer for this. I've tried every way I know and still haven't succeeded. Could you please point me to a way to configure the Ultimaker 3S for use with the Prusa slicer? Thanks for all the help!

GregValiant, 2023-03-06:
I'm sorry but I'm not an UltiMaker owner.  My printer is an Ender 3 Pro.
I would think the problem is the "Gcode Flavor" as your S3 uses Griffon which may not be supported by PrusaSlicer.
@MariMakes might know(?) as I'm sure there is some "keep up with the competition" research that goes on.

5axes, 2023-03-14:
> You seem to have a bad STL file there. I tried to open it in Mesh Mixer, IdeaMaker, PrusaSlicer, and MS 3D Builder. They all reported that it is an invalid file.

It's a 3MF file , whithout any issue, size 80x80x23.8 
[image]

MariMakes, 2023-03-14:
Hey @Mkanton,

I don't believe that the slicer is the issue but the way you generated the model. 
What CAD program do you use? Solidworks, Fusion360, OpenSCAD, Inventor, or something else? 

Does it have an option to export to .3mf? That way your slicer recognizes the size and position of the model .

Mkanton, 2023-03-14:
Hello @MariMakes MariMakes and thanks for replying.

This model was generated in Fusion 360.

You are right! I generated the model in **.3MF** format and was able to print normally.

For some reason, there must have been some error in generating the model and the printing had failed.

Fortunately, everything is ok now.

Thank you very much @MariMakes

Mkanton, 2023-03-14:
> > Você parece ter um arquivo STL ruim lá. Eu tentei abri-lo no Mesh Mixer, IdeaMaker, PrusaSlicer e MS 3D Builder. Todos eles relataram que é um arquivo inválido.
> 
> É um arquivo 3MF, sem qualquer problema, tamanho 80x80x23.8 [image]

Thanks @GregValiant 

I generated the model again in .3MF format and was able to print normally.

mehrdadhp63, 2025-01-29:
Hello dear friends. I find a solution for this problem. All little object have this problem. You should make a cube in 1×1×1 mm far from your object. Cura add two objects together and think thats a big object and dont change the size. You can add other liitle object you want print and this multy object can work as that mini cube too.