Ultimaker/Cura #14780: Creality CR-10 slices fine, but CR-10 Smart has an unexpected error

Roger559, 2023-03-05:
### Application Version

5.3.0-beta.2

### Platform

Windows 10

### Printer

See title

### Reproduction steps

Sliced part with CR-10 profile with success. 
Sliced part with CR-10 Smart profile with error.

This part is in "Generic PLA."

Tried again with "Generic PETG" with success for **both** profiles.

There seems to be a problem in the profile for Generic PLA, but not for Generic PETG.

### Actual results

I am still able to print PLA with the older CR-10 profile on my CR-10 Smart.

### Expected results

I am still able to print PLA with the older CR-10 profile on my CR-10 Smart.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10892726/cura.log)
[CCR10S_Blade-tipL.zip](https://github.com/Ultimaker/Cura/files/10892729/CCR10S_Blade-tipL.zip)

GregValiant, 2023-03-06:
Thanks for the report and the file.
I don't think this has anything to do with the definition file in use.  There is a slicing bug that can keep a model from slicing in certain orientations on the Cura build plate.  I rotated the model 90°CCW and it sliced.  Sometimes moving the model will fool the bug as well.
I'll mark this as a duplicate slicing error and someone from the Cura team will take a look.  (FYI This bug was worked on for the 5.3beta release but it's obvious that it requires more work to get rid of.)

Roger559, 2023-03-06:
Thanks!  I'll try rotating the model.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker