Ultimaker/Cura #14789: Conical support missing

Chad541121, 2023-03-06:
### Application Version

5.3.0 beta 2

### Platform

Window 10

### Printer

Ender 5 Plus

### Reproduction steps

1. When conical support is enabled, sliced.  There is no support shown in the preview.
2. If I use one of my custom profiles from Cura 5.0.0 then the conical support is shown, however there is no brim.
3. 

### Actual results

1. There is no support shown in the preview.
2. There is no brim.

### Expected results

1. Should see conical support with brim
2. Should see brim

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Problem_No conical support.zip](https://github.com/Ultimaker/Cura/files/10899427/Problem_No.conical.support.zip)

GregValiant, 2023-03-06:
Thanks for the report and the file.
I can confirm that Conical Supports is broken for certain printers.  When my Ender 3 Pro is active it simply isn't available and for some Creality and Anycubic printers it might be available but it just doesn't work.
#14666 , #14752, and a couple of others appear to be duplicates of this.
I'll mark it as a duplicate and the Cura team will take a look.

MariMakes, 2023-03-20:
Hey @Chad541121,

Thanks for your report 👍
We were able to resolve the problem with Conical Support in 5.3.0 Stable. 

I can successfully slice your model. You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0
[image]

I'll close this issue since there is a solution in place. 
Please let us know if you run into any new issues. 💪

closed 2023-03-20 by MariMakes

Chad541121, 2023-03-21:
The problem is not fixed!!!

 

Add support brim and you can see that it does not generate support, and first layer doesn’t touch the build plate!  Change support brim to zero and you will see the support touches the build plate.

 

 <https://user-images.githubusercontent.com/40423138/226412952-bb752a08-7c45-44d9-b31d-765d5471285d.png>

Chad541121, 2023-03-21:
Apologies.  I figured out that my saved profiles don’t update.  Is there a way to update my existing profiles, or do I have to rebuild all my profiles?

 

 <https://user-images.githubusercontent.com/40423138/226412952-bb752a08-7c45-44d9-b31d-765d5471285d.png>

Chad541121, 2023-03-21:
There is a setting somewhere that breaks the support brim feature.  I just changed all my settings to match my old profile and now the support brim disappears again.

 

 <https://user-images.githubusercontent.com/40423138/226412952-bb752a08-7c45-44d9-b31d-765d5471285d.png>

Chad541121, 2023-03-21:
> Hey @Chad541121,
> Could it be related to #14666 (comment)?
> In some dual extrusion printjobs we see small brims if the Buildplate Adhesion Extruder is not defined.
> If you define it to be one of your extruders instead of overridden it should work as intended.

I do not have a dual extender. How would I define it. How is build plate adhesion for the model relevant to support brim? I would think they should be separate entities. Chad's iPhone 14 Pro

Chad541121, 2023-03-22:
I only have 1 extruder and those options aren’t available in the 5.3.0 that I downloaded.  Even if I search for extruder, the options don’t show up.

 

 <https://user-images.githubusercontent.com/40423138/226633405-4eff4917-c120-4172-9bf6-aa4403e1dc73.png>

MariMakes, 2023-03-28:
Hey @Chad541121,

Apologies, I must have confused your report for a different one. 
I took another look at your original project file. 
I see that a support brim is generated, but only the inside, this is expected. 
That's also what is described in the tooltip: 
[image]

[image]

If you want a bigger brim on the support, you'll have to turn on the brim for the model and make sure to uncheck the Brim Only on Outside. Than it will look like this: 
[image]

Is that what you are looking for?

Chad541121, 2023-03-28:
When I enable support brim it does nothing. No support brim is generated unless model support is also enabled. I want 3-4 layers of support brim and zero model support brim. If model support is not enabled. I’m then support brim leaves a gap between print bed and support instead of adding support brim it actually disappearsChad's iPhone 14 Pro

MariMakes, 2023-04-03:
This discussion seems to have resulted in the following ticket: 
https://github.com/Ultimaker/Cura/issues/14983